saki4510t/UVCCamera #96: Missing com.serenegiant.utils in the public repository

derchoff, 2016-12-22:
I've cloned the last repository version from **https://github.com/saki4510t/UVCCamera.git**

I've opened the project with the Android studio and i've builded it.

I've tryed the library in my app and it crash in the **USBMonitor creator** on `mAsyncHandler = HandlerThreadHandler.createHandler(TAG);`

It cannot instanciate the HandlerThreadHandler because it seems that the **com.serenegiant.utils** is missing.

Before today i was using an "old" repository version ( cloned on october 2016 ) and my app  worked well. 

Does the missing com.serenegiant.utils is in another github project  ?

saki4510t, 2016-12-22:
Hi,
I'm not sure how you import and build, it seems your project can't resolve dependancies correctly.
Your top level `build.gradle` should include

```
maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
```
in `repositories` in `allprojects` section and `build.gradle` of `usbCameraCommon` module should include
```
compile("com.serenegiant:common:${commonLibVersion}") {
	exclude module: 'support-v4'
}
```
(`commonLibVersion` is defined in top level `build.gradle` file as `1.1.9`, latest one is `1.3.1` though).

Then sync your gradle project, gradle will automatically download it from Github.

derchoff, 2016-12-23:
Hi, 

Thank you for your answer. 

This is my first android application and my first java application so many things still not obvious. 

Should i add the section in the build.gradle files of my app where i use the libuvccamera-debug.aar  ?

I open the UVCCamera project with Android Studio and i build it. 
In the "**UVCCamera\libuvccamera\build\intermediates\exploded-aar\com.serenegiant\common\1.1.9\jars**" folder there is a **classes.jar** file and into this file there are the com.serenegiant.utils classes.
But in the **UVCCamera\libuvccamera\build\outputs\aar\libuvccamera-debug.aar** these classes are missing. 
Even after a "Clean Project" (i've checked that all the build folder and files are cancelled)  followed by a "Make Project".

I've checked the UVCCamera project and in the build.gradle file i see that : 

```
// Top-level build file where you can add configuration options common to all sub-projects/modules.
buildscript {
    repositories {
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'
    }
}

allprojects {
    repositories {
		maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }
}

task clean(type: Delete) {
	delete rootProject.buildDir
}

ext {
	supportLibVersion = '25.0.0'  // variable that can be referenced to keep support libs consistent
	commonLibVersion= '1.1.9'
	versionBuildTool = '24.0.3'
	versionCompiler = 25
	versionTarget = 23
	versionNameString = '1.0.0'
	javaSourceCompatibility = JavaVersion.VERSION_1_7
	javaTargetCompatibility = JavaVersion.VERSION_1_7
}
```
It seems correct. Should i update the commonLibVersion to 1.3.1 ? 

In the build.gradle of usbCameraCommonProject i've : 

```
apply plugin: 'com.android.library'

android {
    compileSdkVersion versionCompiler
   	buildToolsVersion versionBuildTool

   	compileOptions {
   		sourceCompatibility javaSourceCompatibility
   		targetCompatibility javaTargetCompatibility
   	}

    defaultConfig {
        minSdkVersion 18
        targetSdkVersion versionTarget

    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    compile("com.serenegiant:common:${commonLibVersion}") {
   		exclude module: 'support-v4'
   	}
    compile project(':libuvccamera')
}
```

MikeMijnheer, 2017-05-15:
@derchoff did you manage to solve this? I also miss the 'utils' in the repository.

xiaochongdeweiba, 2017-09-28:
you can use android studio goto Declaration,then get all used files in utils.

twoputt, 2018-06-11:
Hi @derchoff, I'm also experiencing this issue and wondering if you ever solved it.

twoputt, 2018-06-11:
Oh!  @derchoff , the sources are at:
    https://github.com/saki4510t/libcommon/tree/master/common/src/main/java/com/serenegiant/utils

moriwone, 2019-04-21:
hi, I sync fail 
ERROR: Failed to resolve: com.serenegiant:common:1.5.20  
My build.gradle file as :
buildscript {
    repositories {
        google()
        jcenter()
    }
    dependencies {
        // gradle插件版本
        classpath 'com.android.tools.build:gradle:3.1.4'
        classpath 'com.github.dcendents:android-maven-gradle-plugin:2.0'
    }
}

allprojects {
    repositories {
        google()
        maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }
}

task clean(type: Delete) {
    delete rootProject.buildDir
}

ext {
    supportLibVersion = '25.3.1'  // variable that can be referenced to keep support libs consistent
    commonLibVersion= '1.5.20'
    //versionBuildTool = '26.0.2'
    //versionCompiler = 25
    //versionTarget = 22
    //versionNameString = '1.0.0'
    javaSourceCompatibility = JavaVersion.VERSION_1_7
    javaTargetCompatibility = JavaVersion.VERSION_1_7
}

lijianchen, 2019-09-16:
> Hi,
> I'm not sure how you import and build, it seems your project can't resolve dependancies correctly.
> Your top level `build.gradle` should include
>
> ```
> maven { url [...]

The correct maven url for common library is "https://" prefix instead of "http://" as below in build.gradle: 

allprojects {
repositories {

              maven { url 'https://raw.github.com/saki4510t/libcommon/master/repository/' }
              google()
              jcenter()
}
}

SnippyValson, 2021-01-28:
@derchoff Did you manage to get UVCCamera working?

miqt, 2021-06-15:
结合以上，我把： maven { url 'https://raw.github.com/saki4510t/libcommon/master/repository/' }
修改为：maven { url 'https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/' }

解决了我的问题。

并且，我发现由于墙的原因，我必须连接VPN才能访问：https://raw.githubusercontent.com/saki4510t/libcommon/master/repository/com/serenegiant/common/2.12.4/common-2.12.4.pom