saki4510t/UVCCamera #97: UVCCamera Video full screen stretching

oy-paddy, 2016-12-27:
hi,saki4510t:

I find onMeasure void (width ,height) is  Mobile phone wide high,Would lead
Video full screen stretching！

Look forward to your reply！Thank you.

oy-paddy, 2016-12-27:
I solve the problem! in com.serenegiant.widget.UVCCameraTextureView class ,the video is full screen not stretching .Here is the code：

    @Override
    protected void onMeasure(int widthMeasureSpec, int heightMeasureSpec) {

//		if (mRequestedAspect > 0) {
//			int initialWidth = MeasureSpec.getSize(widthMeasureSpec);
//			int initialHeight = MeasureSpec.getSize(heightMeasureSpec);
//
//			final int horizPadding = getPaddingLeft() + getPaddingRight();
//			final int vertPadding = getPaddingTop() + getPaddingBottom();
//			initialWidth -= horizPadding;
//			initialHeight -= vertPadding;
//
//			final double viewAspectRatio = (double)initialWidth / initialHeight;
//			final double aspectDiff = mRequestedAspect / viewAspectRatio - 1;
//
//			if (Math.abs(aspectDiff) > 0.01) {
//				if (aspectDiff > 0) {
//					// width priority decision
//					initialHeight = (int) (initialWidth / mRequestedAspect);
//				} else {
//					// height priority decison
//					initialWidth = (int) (initialHeight * mRequestedAspect);
//				}
//				initialWidth += horizPadding;
//				initialHeight += vertPadding;
//				widthMeasureSpec = MeasureSpec.makeMeasureSpec(initialWidth, MeasureSpec.EXACTLY);
//				heightMeasureSpec = MeasureSpec.makeMeasureSpec(initialHeight, MeasureSpec.EXACTLY);
//			}
//		}

        super.onMeasure(widthMeasureSpec, heightMeasureSpec);
    }

closed 2016-12-27 by oy-paddy

Maggie-Wan, 2019-06-14:
Thank you for your sharing.
That helps me solve my problem.
But I found the onMeasure method incom.serenegiant.widget.AspectRatioTextureView class.