saki4510t/UVCCamera #100: UVC camera library fails with android 7

thisum, 2016-12-29:
Hello saki4510t,

I've been testing UVCCamera library with android 7. I tested all the basic samples (usbCameraTest0, usbCameraTest and usbCameraTest2) on  a google pixel phone. But unfortunately none of these samples are working. And the error I'm getting is;

```
java.lang.IllegalArgumentException: Failed to set preview size
at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:306)
at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:277)
at com.serenegiant.usbcameratest2.MainActivity$3$1.run(MainActivity.java:191)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
at java.lang.Thread.run(Thread.java:761)

```

in the onConnect() method, I tried with different height-with combinations, but it kept failing to set the previewSize() of the camera successfully.  Meantime I also noticed that "supportedSize" of the camera gives, null value. 

Could you please have a look at it?

Thanks.

saki4510t, 2017-01-19:
Sorry for delay.

> supportedSize" of the camera gives, null value.
this means app can't open/access camera correctly. There are many reasons, for example,
* Some Android devices need USB hub between camera and device.
* Cabling issue; cheap one, longer cable will many times cause trouble.
* Does you camera is UVC camera? this repository only supports UVC cameras.
* I don't have pixel phone, Google may change something to reject UVC camera.
   Did you see messages that kernel driver detached on logCat? usually they come from EventHub when app try to open/access camera. It looks something like this;
    ```
    I/EventHub: Removing device HD Pro Webcam C920 due to epoll hang-up event.
    I/EventHub: Removed device: path=/dev/input/event7 name=HD Pro Webcam C920 id=103 fd=206 classes=0x80000001
    I/EventHub: Removing device '/dev/input/event7' due to inotify event
     ```
    These messages came from Android system so you will need to un-filter logCat.

saki

mashhur, 2017-06-05:
Hi saki4510t,
I have been using your source and my camera UVC type but couldn't get worked.

Can you please take a look following logs, you might realize the problem. Thanks for your time consideration.

If you take a look at following log, **uvc_get_stream_ctrl_format_size_fps** (of UVCPreview.cpp -> prepare_preview when thread created.) gives always error (-51) and that's why the Rectangle/Preview size is ZERO.
 - Camera found, got info
 - Camera opened (handler created)
 - Preview failed (UVCCamera->setPreviewSize() gives error from uvc driver)

PLZ HELP ME HERE!
------------------------------------------------------------------

`**06-04 19:11:34.032 12672-12721/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/003,desc=83,busnum=1,devnum=3,rawDesc=[B@66c73f1**
06-04 19:11:34.032 12672-12721/com.serenegiant.usbcameratest0 V/GC6500_Test0: onConnect:
06-04 19:11:34.050 12672-12705/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
06-04 19:11:34.051 12672-12705/com.serenegiant.usbcameratest0 I/libusb/core: [12705*core.c:2085:libusb_init2]:libusb v1.0.19.10903
06-04 19:11:34.051 12672-12705/com.serenegiant.usbcameratest0 I/libusb/usbfs: [12705*android_usbfs.c:504:op_init2]:start up hotplug event handler
06-04 19:11:34.051 12672-12705/com.serenegiant.usbcameratest0 I/libusb/usbfs: [12705*android_usbfs.c:512:op_init2]:call android_scan_devices
06-04 19:11:34.058 12672-12705/com.serenegiant.usbcameratest0 V/GC6500_Test0: open camera:0
06-04 19:11:34.058 12672-12705/com.serenegiant.usbcameratest0 V/GC6500_Test0: supportedSize:{"formats":[{}]}
06-04 19:11:34.064 12672-12705/com.serenegiant.usbcameratest0 V/GC6500_Test0: setPreviewSize: 2
06-04 19:11:34.064 12672-13819/com.serenegiant.usbcameratest0 E/libUVCCamera: **[13819*UVCPreview.cpp:508:prepare_preview]:could not negotiate with camera:err=-51
06-04 19:11:34.071 12672-12705/com.serenegiant.usbcameratest0 V/GC6500_Test0: setPreviewDisplay: 0**
06-04 19:11:34.077 12672-12705/com.serenegiant.usbcameratest0 D/InputTransport: Input channel constructed: fd=104
06-04 19:11:34.077 12672-12705/com.serenegiant.usbcameratest0 D/ViewRootImpl@4989e57[Toast]: setView = android.widget.LinearLayout{3841d44 V.E...... ......I. 0,0-0,0} touchMode=true
06-04 19:11:34.100 12672-12724/com.serenegiant.usbcameratest0 D/mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [442x132]-format:1
**06-04 19:11:34.100 12672-12705/com.serenegiant.usbcameratest0 D/ViewRootImpl@4989e57[Toast]: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1**
06-04 19:11:34.677 12672-12672/com.serenegiant.usbcameratest0 D/ViewRootImpl@481f269[MainActivity]: ViewPostImeInputStage processPointer 0
06-04 19:11:34.687 12672-12705/com.serenegiant.usbcameratest0 D/InputTransport: Input channel constructed: fd=105
06-04 19:11:34.687 12672-12705/com.serenegiant.usbcameratest0 D/ViewRootImpl@6c35e2d[Toast]: setView = android.widget.LinearLayout{baeff62 V.E...... ......I. `

usereman, 2017-07-31:
did you got any solution to this?

109021017, 2018-01-11:
Did you got any solution to this?

along1991, 2019-01-17:
How can I solve it?

along1991, 2019-01-17:
> Did you got any solution to this? @mashhur

How can I solve it?

namputo, 2019-06-06:
hi @saki4510t I face a similar issue with POCOF1 mobile running Android Oreo. The problem is that it is not consistent with all the phones i have (I tested with 5 different POCO phones). Amongst these, 3 of them did not work. I saw some third party external cameras worked with out any problem with the same hardware set up. Any clue on what is happening will be really useful

wade-fs, 2020-03-16:
Could any one show me how to set preview size lower than 640x480 for usbCameraTest7?
I tried to modified DEFAULT_PREVIEW_WIDTH + DEFAULT_PREVIEW_HEIGHT in UVCCamera/UVCPreview.h, but it is of no use.
I think the reason we can't turn on both cameras at the same time is caused by the default resolution.