saki4510t/UVCCamera #103: preview show  current timestamp

oy-paddy, 2017-01-03:
Hi,saki4510t:
I want to show the current timestamp ,How to let the video and preview display the current timestamp.
Looking forward to your reply! thank you.

saki4510t, 2017-02-18:
Hi, sorry for delayed.

Well what timestamp and where do you want to show?

oy-paddy, 2017-02-20:
Hi saki4510t !
Thank you for your replyPlay the video to show the current time of the watermark, playback video can also be recorded when the time to see the watermark, the details of the attached picture, displayed in the lower right or upper right or left or left.

xwpcom, 2017-05-02:
It is useful to have OSD for both preview and capture.
OSD can be time/custom text and so on

jp1017, 2017-05-03:
@xwpcom can you show the demo? :+1:

oy-paddy, 2017-05-03:
this is a good idea! Hope to join the development plan!

jp1017, 2017-06-29:
@saki4510t haha, any idea for this? 

@oy-paddy Have you make it? :smile:

dinhtung249, 2022-01-25:
Can you share your solve?
Many thank!