saki4510t/UVCCamera #109: Set Preview Size Error And nativeSetPreviewSize() method Returns -51

zhanghaien, 2017-01-21:
Hello saki4510t,

I've been testing UVCCamera library with some  android devices that  have USB OTB support. I tested all the basic samples (usbCameraTest0, usbCameraTest and usbCameraTest2). But unfortunately none of these samples are working. And the error I'm getting from usbCameraTest2 is：
      
01-21 15:08:47.840 22669-22684/? W/System.err: java.lang.IllegalArgumentException: Failed to set preview size
01-21 15:08:47.840 22669-22684/? W/System.err:     at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:333)
01-21 15:08:47.840 22669-22684/? W/System.err:     at com.serenegiant.usb.UVCCamera.setPreviewSize(UVCCamera.java:303)
01-21 15:08:47.840 22669-22684/? W/System.err:     at com.serenegiant.usbcameratest2.MainActivity$3$1.run(MainActivity.java:201)
01-21 15:08:47.840 22669-22684/? W/System.err:     at android.os.Handler.handleCallback(Handler.java)
01-21 15:08:47.840 22669-22684/? W/System.err:     at android.os.Handler.dispatchMessage(Handler.java)
01-21 15:08:47.840 22669-22684/? W/System.err:     at android.os.Looper.loop(Looper.java)
01-21 15:08:47.840 22669-22684/? W/System.err:     at android.os.HandlerThread.run(HandlerThread.java)

01-21 15:08:47.840 22669-1029/? E/libusb/usbfs: [1029*android_usbfs.c:2876:handle_control_completion]:LIBUSB_TRANSFER_STALL
01-21 15:08:47.840 22669-22684/? E/libuvc/stream: [22684*stream.c:391:_prepare_stream_ctrl]:uvc_query_stream_ctrl:UVC_GET_MIN:err=-9
01-21 15:08:47.840 22669-22684/? E/libusb/usbfs: [22684*android_usbfs.c:2876:handle_control_completion]:LIBUSB_TRANSFER_STALL
01-21 15:08:47.840 22669-22684/? E/libuvc/stream: [22684*stream.c:391:_prepare_stream_ctrl]:uvc_query_stream_ctrl:UVC_GET_MIN:err=-9
01-21 15:08:47.840 22669-22684/? E/libuvc/stream: [22684*stream.c:427:_uvc_get_stream_ctrl_format]:_prepare_stream_ctrl:err=-9

but in the nativeGetSupportedSize(final long id_camera) method, I got the message({"formats":[{"index":1,"type":6,"default":1,"size":["1280x720"]}]}) from uvcCamera device  and set the preview size to 1280 * 720, but it kept failing to set the previewSize() of the camera successfully. Meantime nativeSetPreviewSize() method returns -51 ,not 0.

Could you please have a look at it?

Thanks.

jp1017, 2017-02-12:
Use other phones maybe.

AloyASen, 2017-03-01:
This error might occur when the width and height of the preview is set wrongly, i.e, width in the place of height and vice versa, that actually results in an abnormal preview size which might not be supported and hence ends up throwing an error.