saki4510t/UVCCamera #112: Bulk Transfer results in LIBUSB_TRANSFER_TIMED_OUT

ojotoxy, 2017-02-02:
Occasionally upon connecting to the device and opening a stream, I only get LIBUSB_TRANSFER_TIMED_OUT inside _uvc_stream_callback in stream.c in libucv.

I can only make this go away by going back to the home screen and back into the app and reconnecting.

My device has a blinking LED when the transfer is working, and it's not blinking in this state.

Any ideas why this would happen?

saki4510t, 2017-02-03:
Hi,

As I can see sources in `libusb`, it only returns from `usbi_handle_transfer_cancellation()` function in `io.c` when `USBI_TRANSFER_TIMED_OUT` flag is set.  
And the flag is only set in `handle_timeout()` called from `handle_timeouts_locked()` <- `handle_timeouts()` <- `handle_events()` when system call `poll()` is expired(`libuvc` calls `libusb_handle_events` to handle USB events and the timeout value is set to 60 sec in it, I assume shorter timeout value may be better though).  
This means the error itself occurred in kernel side. I assume bulk endpoint on your device may become halt state by some reasons. Could you try to call `libusb_clear_halt` when you receive `USBI_TRANSFER_TIMED_OUT ` on bulk endpoint? (FYI, isochronous endpoint never become halt state)

The reasons will be;

1. connection quality is low / electrically unstable condition... too long cable, cheap cable etc.
2. too large transfer than available bandwidth(bandwidth is also affected by connection quality/hardware).
3. ...

USB interface on Android devices are much sensitive than usual PC and shorter cable/USB 2.0 hub may also help you.

ojotoxy, 2017-02-04:
Thanks for your response!

In _uvc_stream_callback, right after this...

```
        case LIBUSB_TRANSFER_TIMED_OUT:
        case LIBUSB_TRANSFER_STALL:
```

I added
```
 LOGE("Attempting to clear halt.");                                
 libusb_clear_halt(transfer->dev_handle, transfer->endpoint);      

```

Then when the error occurs...

```
02-03 15:52:47.991 6416-6492/com.xyzco.android I/libusb/core: [6492*core.c:2085:libusb_init2]:libusb v1.0.19.10903
02-03 15:52:47.991 6416-6492/com.xyzco.android I/libusb/usbfs: [6492*android_usbfs.c:504:op_init2]:start up hotplug event handler
02-03 15:52:47.991 6416-6492/com.xyzco.android I/libusb/usbfs: [6492*android_usbfs.c:512:op_init2]:call android_scan_devices
02-03 15:52:47.992 6416-6492/com.xyzco.android E/libuvc/device: [6492*device.c:330:uvc_open]:internal_devh->info->ctrl_if.bEndpointAddress is null
02-03 15:52:47.998 6416-7946/com.xyzco.android I/libUVCCamera: [7946*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,360)@YUYV
02-03 15:52:53.005 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.005 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.005 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.006 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.007 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.008 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:53.008 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:53.008 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.010 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.011 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.011 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.011 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.011 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.012 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.012 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.012 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.012 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.013 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.013 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:52:58.013 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:52:58.013 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
[... cut ...]
02-03 15:53:03.021 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:53:03.021 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.
02-03 15:53:03.021 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:1001:_uvc_stream_callback]:retrying transfer, status = 2
02-03 15:53:03.021 6416-7945/com.xyzco.android E/libuvc/stream: [7945*stream.c:998:_uvc_stream_callback]:Attempting to clear halt.

```

That repeats every 5 seconds. 

Seems something else may be wrong :)

ojotoxy, 2017-02-04:
Some other facts:

- Once a connection is working, it (almost) never stops. I left it on overnight several times with no issues.
- Once the connection is broken, disconnecting and reconnecting within the app does not solve the problem. Only going back to the home screen and reopening the app (without quitting, same activity object will be active) reliably make the connection work again.

It all seems to me like there is a race condition somewhere, in libuvc or the java, or the kernel.

saki4510t, 2017-02-18:
As I can see your log, your camera does not have camera control interface. Did you call something camera control function? and does this issue occur when using other usual camera (that has camera control interface)?
```
02-03 15:52:47.992 6416-6492/com.xyzco.android E/libuvc/device: [6492*device.c:330:uvc_open]:internal_devh->info->ctrl_if.bEndpointAddress is null
```

Mr-Grieves, 2018-01-23:
I am dealing with a similar issue.

My situation:
I am using a framegrabber (instead of a webcam) connected to a the DVI output of a computer. The framegrabber supports both USB2 and USB3 output connections. Everything works fine when using the USB2 cable, however, I can only get a frame rate of about 10fps. The framegrabber can output higher FPS using USB3, but when I try to run my app (or the usbCameraTest), it either can't connect at all or it connects briefly, the framegrabber sends one frame, and then stops transmitting.

The error I keep seeing in the log is the following:
`[18336*android_usbfs.c:2656:handle_bul_completion]:LIBUSB_TRANSFER_STALL`

also, I see the error you mentioned previously:
`[18043*device.c:330:uvc_open]:internal_devh->info->ctrl_if.bEndpointAddress is null`

Any suggestions?