saki4510t/UVCCamera #113: Capturing full res stills

Noloxs, 2017-02-03:
Hi.

I have been looking a bit around and I can't find any reference to capturing a still picture that isn't a grab of the preview.

Is there any way of doing this?

saki4510t, 2017-02-18:
Hi,

Do you mean do you want to use still capture function on your camera?
If so, unfortunately current libuvc does not support it and this repository only supports capturing preview image.

Noloxs, 2017-02-18:
Is there then a way of changing the size of the preview just before capturing the frame and then switching back to the lower res preview setting?

Noloxs, 2017-02-20:
We are currently trying to achieve this by:

1. Releasing the preview and starting a new one with the high res settings.
2. Once we have retrieved the image we release the high res preview
3. Start another new preview with the old settings.

This somewhat works but we experience the following error rather frequently, which causes the entire app to crash:

```Java
ERROR: detaching thread with interp frames (count=8)
"CameraThread" prio=5 tid=25 RUNNABLE
   | group="main" sCount=0 dsCount=0 obj=0x43785f68 self=0x6b1714d8
   | sysTid=26169 nice=0 sched=0/0 cgrp=apps handle=1810877992
   | state=R schedstat=( 0 0 0 ) utm=0 stm=0 core=0
     at com.serenegiant.usb.UVCCamera.nativeUpdateGainLimit(Native Method)
     at com.serenegiant.usb.UVCCamera.updateCameraParams(UVCCamera.java:915)
     at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.handleStartPreview(AbstractUVCCameraHandler.java:492)
     at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.handleMessage(AbstractUVCCameraHandler.java:306)
     at com.serenegiant.usbcameracommon.UVCCameraHandler.handleMessage(UVCCameraHandler.java:31)
     at android.os.Handler.dispatchMessage(Handler.java:102)
     at android.os.Looper.loop(Looper.java:136)
     at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.run(AbstractUVCCameraHandler.java:747)
VM aborting
Fatal signal 6 (SIGABRT) at 0x00000918 (code=-6), thread 26169 (CameraThread)
```

So we were hoping that there was a more reliable way of changing the preview size. 
Preferably in a way that did not require us to reinitialize all the elements every time.

laurentquark, 2017-05-01:
if i understand, use the ImageReader

closed 2022-09-29 by Noloxs