saki4510t/UVCCamera #114: UVCCamera4 cann't work on huawei smartphone

szandroid, 2017-02-11:
Hi saki:
After I installed UVCCamera4 apk and open it on huawei KIW-CL100 phone(android 5.1), using UVCCamer+USB-HUB connect to the phone, it is black on the screen.
how to fix, thx.

BAILOOL, 2017-02-11:
I have encountered tons of problems with Huawei phones. At first, I would suggest you to test your UVC device with the other non-Huawei phone to make sure your device is actually working properly.

saki4510t, 2017-02-18:
Hi,

Did your device detect camera and show dialog to request permission to access USB device when you connect your camera?
If not, your device will not support OTG(USB host) function or OTG  function does not work well.

If your device detects camera, did other samples work or not?
If all samples does not work, you would see errors on logCat. Could you share log from the time before connecting camera to after start app w/o filter (if it is long share it on your `gist`)

@BAILOOL Thank you for providing information. I also know HUAWEI devices sometimes have issues on OpenGL|ES and MediaCodec/MediaMuxer, internal camera...

ajrunhaar, 2017-09-08:
Dear Saki,

Firstly, thank you so much for this amazing project.

I too have encountered this problem with Huawei phone, but also with numerous other "low cost" phones running Android 5.x such as the Samsung J1. These devices do support USB OTG and behave the same as the phones that do display the preview. The camera is connected, permission is requested, and the device connects correctly (mNativePtr exists). The only difference is that there is no preview shown.

I'm not sure where to look for the fault, but I cant see any errors in the logs. 

Have you come across this before?

Kind regards,

Anton

netaivgi, 2017-12-06:
Dear Anton,

Did you manage to find a solution eventually?
I'm trying to connect  UVCCamera to honor6x and having the same issue as yours.

Kind regards,

Neta
@ajrunhaar