saki4510t/UVCCamera #115: java.util.concurrent.TimeoutException: android.media.MediaMuxer.finalize() timed out after 10 seconds

jp1017, 2017-02-13:
Record per 10 minutes error of Test4: 

```
java.util.concurrent.TimeoutException: android.media.MediaMuxer.finalize() timed out after 10 seconds
at android.media.MediaMuxer.nativeRelease(Native Method)
at android.media.MediaMuxer.finalize(MediaMuxer.java:240)
at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:187)
at java.lang.Daemons$FinalizerDaemon.run(Daemons.java:170)
at java.lang.Thread.run(Thread.java:841)
```

saki4510t, 2017-02-18:
Hi,

I assume this seems MediaMuxer took too long time to write muxed data.
What device do you use? As I know HUAWEI devices sometimes have issue on MediaCodec/MediaMuxer.

Or if your primary storage physically exists on SD card, writing speed of the card may be slow than required. Please try to use faster one.

jp1017, 2017-02-21:
Hi, @saki4510t 

It's our own device, CPU info here: 

[image]

May be this reason? CPU? :smile: