saki4510t/UVCCamera #128: The Devices can Detect camera but can not get the view!

codeinging, 2017-03-09:
I encounter the problem.My Test Devices such as RedMi 1s.
The logcat:
03-09 18:36:09.895 7932-7932/com.serenegiant.usbcameratest0 I/View: Touch down dispatch to android.widget.Button{422385e0 VFED..C. ........ 414,0-620,96 #1020019 android:id/button1}, event = MotionEvent { action=ACTION_DOWN, id[0]=0, x[0]=102.21362, y[0]=79.316956, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=14451903, downTime=14451903, deviceId=3, source=0x1002 }
03-09 18:36:09.914 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: create handle(0x63d0f850) (w:688, h:484, f:1)
03-09 18:36:09.918 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 684.00, 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.919 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: clear (0.00, 484 - 484.00 = 0.00, 684.00 - 0.00 = 684.00, 484.00 - 0.00 = 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.923 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x63d0f460>
03-09 18:36:09.973 7932-7932/com.serenegiant.usbcameratest0 I/View: Touch up dispatch to android.widget.Button{422385e0 VFED..C. ...P.... 414,0-620,96 #1020019 android:id/button1}, event = MotionEvent { action=ACTION_UP, id[0]=0, x[0]=102.21362, y[0]=79.316956, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=14451983, downTime=14451903, deviceId=3, source=0x1002 }
03-09 18:36:09.974 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings: get setting for user 0 by user 0 so skipping cache
03-09 18:36:09.974 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings:  from settings cache , name = sound_effects_enabled , value = 0
03-09 18:36:09.978 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: create handle(0x61debfa8) (w:688, h:484, f:1)
03-09 18:36:09.982 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 684.00, 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.982 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: clear (0.00, 484 - 484.00 = 0.00, 684.00 - 0.00 = 684.00, 484.00 - 0.00 = 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.984 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x63d0f460>
03-09 18:36:10.008 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Flushing caches (mode 0)
03-09 18:36:10.009 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d0f360) (w:688 h:484 f:1)
03-09 18:36:10.010 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d39ee0) (w:688 h:484 f:1)
03-09 18:36:10.011 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d0f850) (w:688 h:484 f:1)
03-09 18:36:10.012 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x61debfa8) (w:688 h:484 f:1)
03-09 18:36:10.022 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-AM_ON_PAUSE_CALLED ActivityRecord{41fadc30 token=android.os.BinderProxy@41fad290 {com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity}}
03-09 18:36:10.039 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-PAUSE_ACTIVITY handled : 1 / android.os.BinderProxy@41fad290
03-09 18:36:10.414 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-STOP_ACTIVITY_SHOW handled : 0 / android.os.BinderProxy@41fad290
03-09 18:36:17.639 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings: get setting for user 0 by user 0 so skipping cache
03-09 18:36:17.639 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings:  from settings cache , name = access_control_lock_enabled , value = null
03-09 18:36:17.640 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-AM_ON_RESUME_CALLED ActivityRecord{41fadc30 token=android.os.BinderProxy@41fad290 {com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity}}
03-09 18:36:17.642 7932-7932/com.serenegiant.usbcameratest0 V/PhoneWindow: DecorView setVisiblity: visibility = 0 ,Parent =ViewRoot{421b9618 com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity,ident = 0}, this =com.android.internal.policy.impl.PhoneWindow$DecorView{42042b48 V.E..... R....... 0,0-720,1280}
03-09 18:36:17.660 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-RESUME_ACTIVITY handled : 0 / android.os.BinderProxy@41fad290
03-09 18:36:17.660 7932-7932/com.serenegiant.usbcameratest0 V/InputMethodManager: onWindowFocus: null softInputMode=32 first=false flags=#41810100
03-09 18:36:17.664 7932-7932/com.serenegiant.usbcameratest0 I/SurfaceView: updateWindow -- OnPreDrawListener, mHaveFrame = true
03-09 18:36:17.666 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 720.00, 1280.00) opaque 1 <0x6195acf8>
03-09 18:36:17.667 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x6195acf8>
03-09 18:36:17.690 7932-7956/com.serenegiant.usbcameratest0 V/USBMonitor: processConnect:device=UsbDevice[mName=/dev/bus/usb/001/028,mVendorId=7758,mProductId=265,mClass=239,mSubclass=2,mProtocol=1,mInterfaces=[Landroid.os.Parcelable;@422d9508]
03-09 18:36:17.696 7932-7956/com.serenegiant.usbcameratest0 I/dalvikvm: Could not find method android.hardware.usb.UsbInterface.getAlternateSetting, referenced from method com.serenegiant.usb.USBMonitor$UsbControlBlock.getInterface
03-09 18:36:17.696 7932-7956/com.serenegiant.usbcameratest0 W/dalvikvm: VFY: unable to resolve virtual method 770: Landroid/hardware/usb/UsbInterface;.getAlternateSetting ()I
03-09 18:36:17.696 7932-7956/com.serenegiant.usbcameratest0 D/dalvikvm: VFY: replacing opcode 0x6e at 0x005e
03-09 18:36:17.701 7932-7956/com.serenegiant.usbcameratest0 I/USBMonitor: UsbControlBlock:constructor
03-09 18:36:17.710 7932-7956/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
03-09 18:36:17.711 7932-7956/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/028,desc=75,busnum=1,devnum=28,rawDesc=[B@422dcea8
03-09 18:36:17.711 7932-7956/com.serenegiant.usbcameratest0 V/MainActivity: onConnect:
03-09 18:36:17.714 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest0-2/libjpeg-turbo1500.so 0x41fb3f08
03-09 18:36:17.715 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest0-2/libjpeg-turbo1500.so 0x41fb3f08
03-09 18:36:17.715 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest0-2/libjpeg-turbo1500.so 0x41fb3f08, skipping init
03-09 18:36:17.716 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest0-2/libusb100.so 0x41fb3f08
03-09 18:36:17.717 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest0-2/libusb100.so 0x41fb3f08
03-09 18:36:17.717 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest0-2/libusb100.so 0x41fb3f08, skipping init
03-09 18:36:17.718 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest0-2/libuvc.so 0x41fb3f08
03-09 18:36:17.719 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest0-2/libuvc.so 0x41fb3f08
03-09 18:36:17.719 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest0-2/libuvc.so 0x41fb3f08, skipping init
03-09 18:36:17.719 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest0-2/libUVCCamera.so 0x41fb3f08
03-09 18:36:17.721 7932-7954/com.serenegiant.usbcameratest0 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest0-2/libUVCCamera.so 0x41fb3f08
03-09 18:36:17.736 7932-7954/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
03-09 18:36:17.736 7932-7954/com.serenegiant.usbcameratest0 I/libusb/core: [7954*core.c:2085:libusb_init2]:libusb v1.0.19.10903
03-09 18:36:17.736 7932-7954/com.serenegiant.usbcameratest0 I/libusb/usbfs: [7954*android_usbfs.c:504:op_init2]:start up hotplug event handler
03-09 18:36:17.736 7932-7954/com.serenegiant.usbcameratest0 I/libusb/usbfs: [7954*android_usbfs.c:512:op_init2]:call android_scan_devices
03-09 18:36:17.738 7932-7954/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}
03-09 18:36:17.757 7932-8119/com.serenegiant.usbcameratest0 I/libUVCCamera: [8119*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
03-09 18:36:18.415 7932-8127/com.serenegiant.usbcameratest0 D/dalvikvm: threadid=15: interp stack at 0x63f06000
Thanks

sascha-hack, 2017-08-14:
Did you ever find a solution ?

I get the same error ?

`08-13 23:39:25.769 16572-16586/com.serenegiant.usbcameratest0 I/dalvikvm: Could not find method android.hardware.usb.UsbInterface.getAlternateSetting, referenced from method com.serenegiant.usb.USBMonitor$UsbControlBlock.getInterface
08-13 23:39:25.769 16572-16586/com.serenegiant.usbcameratest0 W/dalvikvm: VFY: unable to resolve virtual method 728: Landroid/hardware/usb/UsbInterface;.getAlternateSetting ()I
08-13 23:39:25.799 16572-16586/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/003,desc=69,busnum=1,devnum=3,rawDesc=[B@21b53c98
08-13 23:39:25.819 16572-16585/com.serenegiant.usbcameratest0 I/libusb/core: [16585*core.c:2085:libusb_init2]:libusb v1.0.19.10903
08-13 23:39:25.819 16572-16585/com.serenegiant.usbcameratest0 I/libusb/usbfs: [16585*android_usbfs.c:504:op_init2]:start up hotplug event handler
08-13 23:39:25.819 16572-16585/com.serenegiant.usbcameratest0 I/libusb/usbfs: [16585*android_usbfs.c:512:op_init2]:call android_scan_devices`

sappho192, 2017-08-14:
@sascha-hack 
`08-13 23:39:25.799 16572-16586/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/003,desc=69,busnum=1,devnum=3,rawDesc=[B@21b53c98`
means it seems to be connected with the camera.
Check those links to find the clues https://github.com/saki4510t/UVCCamera/issues/181 https://github.com/saki4510t/UVCCamera/pull/175