saki4510t/UVCCamera #129: Capture images simultaneously?

sammy3d, 2017-03-10:
The usb camera test 6 is working great with a dual lens camera and Galaxy Note 4. I can preview both cameras with no problem. I can also capture an image but only from one lens/camera at a time. I am not trying to capture video yet just images ( ".jpg").

I see the switch statement _toggles_ the `(view.getId())` for the long click method. Is their a way to capture `_camera_view_L_` and `_camera_view_R_` at the same time and get two jpegs? 

@saki4510t  UVCCamera is Great! Many uses. Thank you.

tairtec, 2017-03-11:
Maybe you should try test 7

sammy3d, 2017-03-13:
Test 7 is also set up for two cameras. It does not have any capture methods like test 6 (media muxer etc.).

WildOrangutan, 2017-05-15:
I basically modified test app 7, made one common button and changed picture capture (beside other stuff) like this:

`final File outputFileL = MediaMuxerWrapper.getCaptureFile(Environment.DIRECTORY_DCIM + "_L", ".png");
final File outputFileR = MediaMuxerWrapper.getCaptureFile(Environment.DIRECTORY_DCIM + "_R", ".png");
mHandlerL.captureStill(outputFileL.toString());
mHandlerR.captureStill(outputFileR.toString());`

abneet07, 2018-04-19:
@WildOrangutan thanks you for information. I'm over here trying to capture High resolution Image from Dual Camera at the same time. Do you have any idea what changes i need to make. Any leads will be appreciated.

WildOrangutan, 2018-04-20:
@abneet07 here you go [gist click](https://gist.github.com/WildOrangutan/6afa081bd8c6b76d836590fb2f841210). This is what I've used for some quick testing purposes. It may not be perfect but it does the job.

lgstud-io, 2020-05-07:
> I basically modified test app 7, made one common button and changed picture capture (beside other stuff) like this:
>
> ```
> final File outputFileL = [...]

Code like this does not work.
I have created a new issue about this. 
Whenever I call 2 or more captureStill on more handlers, only the last picture gets saved( because the first image is rewritten all the time and no new images are saved) 
There is something very wrong with this still capturing and I have no idea how to solve this.

heaveninfosoft, 2020-06-28:
> @abneet07 here you go [gist click](https://gist.github.com/WildOrangutan/6afa081bd8c6b76d836590fb2f841210). This is what I've used for some quick testing purposes. It may not be perfect but it does the job.

your tow camera working??? can you send me code for daul camera which can suppord HD and FUll HD??

heaveninfosoft, 2020-06-28:
@WildOrangutan your code support two camerq?

WildOrangutan, 2020-06-28:
It should, since I used it for dual camera.

For anything else, I can't really help you guys. I've used this library for a few days, so I'm definitely not an expert here.

heaveninfosoft, 2020-06-28:
@WildOrangutan but in my it can work can you send me your source code