saki4510t/UVCCamera #130: No sdkHandler field in LibraryPlugin after updating to build tools 2.3.0

tairtec, 2017-03-11:
In libuvccamera module build.gradle

```
def ndkBuildingDir = project.plugins.findPlugin('com.android.library').sdkHandler.getNdkFolder().absolutePath
```

Ater updating com.android.tools.build:gradle to 2.3 the field **sdkHandler** seems to have been renamed on removed

> Error:No such property: sdkHandler for class: com.android.build.gradle.LibraryPlugin

Is there an alternative way to get the NDK directory path in a library module?

saki4510t, 2017-03-12:
You can use following `build.gradle`

```
android {
	compileSdkVersion versionCompiler
	buildToolsVersion versionBuildTool

	compileOptions {
		sourceCompatibility javaSourceCompatibility
		targetCompatibility javaTargetCompatibility
	}

	defaultConfig {
		minSdkVersion 14
		targetSdkVersion versionTarget
		ndk {
			// Specifies the ABI configurations of your native
			// libraries Gradle should build and package with your APK.
			abiFilters 'x86', 'armeabi', 'armeabi-v7a'
		}
	}

	buildTypes {
		release {
			minifyEnabled false
			proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.txt'
		}
	}
	externalNativeBuild {
		ndkBuild {
			path "src/main/jni/Android.mk"
		}
	}
}

dependencies {
	compile fileTree(dir: new File(buildDir, 'libs'), include: '*.jar')
	compile "com.android.support:support-annotations:${supportLibVersion}"
	compile("com.serenegiant:common:${commonLibVersion}") {
		exclude module: 'support-v4'
	}
}
```

But I recommend to use gradle plugin 2.2.3 instead of 2.3 because new NDK support on Android Studio still does not work well (at least NDK support for ndk-build, ex. it does not handle `Application.mk` correctly)

saki4510t, 2017-03-15:
I update `build.gradle` so that it can be use on both Android Studio/gradle plugin 2.3.x and prior Android Studio. Please check latest master branch.

closed 2017-03-15 by saki4510t