saki4510t/UVCCamera #131: How can i get a photo with 1600*1200?

hexilon, 2017-03-12:
Dear saki4510t:
    Now i have UVC camera, is support 1600x1200, then i set PREVIEW_WIDTH = 1600; PREVIEW_HEIGHT = 1200. Because the screen resolution of  my tablet is 1280x800, so when i capture  a photo, it is always less than 1280x800, but i want the photo is 1600x1200, how can i do that? thank you.

laurentquark, 2017-05-01:
don't use the handler code.
set the size yourself in :
camera.setPreviewSize(WIDTH, HEIGHT, UVCCamera.FRAME_FORMAT_MJPEG);
and 
preview surface, the android resize alone for me.

or

use for the receive the frames captured 
mImageReader = ImageReader.newInstance(WIDTH, HEIGHT, PixelFormat.RGBA_8888, 2);
mImageReader.setOnImageAvailableListener(MainActivity.this, null);
and
resize a bitmap in the onImageAvailable() to transfer to the preview

remicoges, 2017-07-04:
@laurentquark Have you achieved this ? I'm trying to get 2048*1536 captures (I set PREVIEW_WIDTH and PREVIEW_HEIGHT to these values) but I get 1080*809 captures