saki4510t/UVCCamera #133: build under Windows

gbaracchino, 2017-03-13:
I'm trying to compile your library on Windows.
I've the following errors:

[armeabi] Install        : libUVCCamera.so => libs/armeabi/libUVCCamera.so
[armeabi] Install        : libjpeg-turbo1500.so => libs/armeabi/libjpeg-turbo1500.so
[armeabi] Install        : libusb100.so => libs/armeabi/libusb100.so
[armeabi] Install        : libuvc.so => libs/armeabi/libuvc.so
[armeabi-v7a] Install        : libUVCCamera.so => libs/armeabi-v7a/libUVCCamera.so
[armeabi-v7a] Install        : libjpeg-turbo1500.so => libs/armeabi-v7a/libjpeg-turbo1500.so
[armeabi-v7a] Install        : libusb100.so => libs/armeabi-v7a/libusb100.so
[armeabi-v7a] Install        : libuvc.so => libs/armeabi-v7a/libuvc.so
[x86] Install        : libUVCCamera.so => libs/x86/libUVCCamera.so
[x86] Install        : libjpeg-turbo1500.so => libs/x86/libjpeg-turbo1500.so
[x86] Install        : libusb100.so => libs/x86/libusb100.so
[x86] Install        : libuvc.so => libs/x86/libuvc.so
[mips] Compile        : jpeg-turbo1500_static <= jsimd_mips_dspr2.S
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5.0/simd/jsimd_mips_dspr2.S:29:263: error: unexpected token, expected comma
.globl jsimd_c_null_convert_mips_dspr2; .align 2; .type jsimd_c_null_convert_mip
s_dspr2, @function; .ent jsimd_c_null_convert_mips_dspr2, 0; jsimd_c_null_conver
t_mips_dspr2: .frame $29, 0, $31; .set push; .set arch=mips32r2; .set noreorder;
 .set noat; .set dspr2;

                      ^
<instantiation>:13:9: error: invalid token in expression
    .if $17 != 0
        ^
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5
.0/simd/jsimd_mips_dspr2.S:41:5: note: while in macro instantiation
    SAVE_REGS_ON_STACK 8, $16, $17
    ^
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5
.0/simd/jsimd_mips_dspr2.S:54:5: error: instruction requires a CPU feature not c
urrently enabled
    lwx $13, $11($6)
    ^
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5
.0/simd/jsimd_mips_dspr2.S:57:5: error: instruction requires a CPU feature not c
urrently enabled
    lwx $13, $12($13)
    ^
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5
.0/simd/jsimd_mips_dspr2.S:96:5: error: instruction requires a CPU feature not c
urrently enabled
    lwx $13, $11($6)
    ^
c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5
.0/simd/jsimd_mips_dspr2.S:99:5: error: instruction requires a CPU feature not c
urrently enabled
    lwx $13, $12($13)
    ^
<instantiation>:10:9: error: invalid token in expression
    .if $17 != 0

... and so on,...

sammy3d, 2017-03-13:
> c:/xx/yy/UVCCamera/UVCCamera/libuvccamera/src/main/jni/libjpeg-turbo-1.5.0/simd/**jsimd_mips_dspr2.S:29:263**: error: unexpected token, expected comma

> 

Getting this same error now but only on the master branch?

tairtec, 2017-03-14:
solution:
libuvccamera/src/main/jni/Application.mk 
> APP_ABI := armeabi armeabi-v7a x86 **mips**

delete **mips**

ravitejachivukula, 2017-03-16:
Removing 'mips' from that line worked like a charm!!

@tairtec it would be great if you could explain what caused the issue and how this specific step resolved it.