saki4510t/UVCCamera #134: How to enable h264 image format

jordanSunco, 2017-03-14:
The UVC camera supports H264 format, but how to enable H264 image format and get H264 output  stream?

saki4510t, 2017-03-16:
Technically, it is possible but unfortunately I don't have plan to add h.264 support on this repository.

The technical basis are available on my blog (sorry, they are Japanese only)
http://serenegiant.com/blog/?p=2932
http://serenegiant.com/blog/?p=2937
http://serenegiant.com/blog/?p=2944
http://serenegiant.com/blog/?p=2956
http://serenegiant.com/blog/?p=3158
...will be continue...

jordanSunco, 2017-03-18:
Thanks，but i can't read Japanese.

fengbenpaao, 2017-07-12:
hi，can you show how to add the h264 patch?  
 1.   I think maybe  I can use FFMPEG     to add the h264 ;
       any idear do you think?
2. my uvc camera use the MJPEG, the preview  will  delay  for  500ms  that video rate:   1280x720；
3. when I change the video rate to 640x360 ， the preview will be ok,  but the  clarity is unacceptable to users;

fengbenpaao, 2017-07-25:
@jordanSunco  do you  hava any idear to do  h264?    my PC Client to show h264 is ok;     but how to add it to libuvccamera  ?

fengbenpaao, 2017-08-08:
@saki4510t  @jordanSunco 

ah, it is so good to finally show my h264 preview;
but there are so much work to do,
there is only one step. go go go!!!!

closed 2017-09-18 by jordanSunco

jinlongxu2018, 2021-06-25:
GOOD,我终于得到了H264视频,并且解决了前面一段时间H264解码后,花屏/绿屏问题.

dinhtung249, 2022-01-25:
> GOOD,我终于得到了H264视频,并且解决了前面一段时间H264解码后,花屏/绿屏问题.

can you share this? tks