saki4510t/UVCCamera #140: Error:Execution failed for task ':libuvccamera:ndkBuild'. > Process 'command '/Users/ajana/Library/Android/sdk/ndk-bundle/ndk-build'' finished with non-zero exit value 2

freebumba27, 2017-03-21:
Android studio issue - `Error:Execution failed for task ':libuvccamera:ndkBuild'.
> Process 'command '/Users/someone/Library/Android/sdk/ndk-bundle/ndk-build'' finished with non-zero exit value 2`

ZahirSoufiane, 2017-03-22:
I don't know if that's good. I went to

C:\Users\Dev\AppData\Local\Android\Sdk\ndk-bundle\ndk-build.cmd

directory and for the ndk-build.cmd i press

Right-Click> Edit and change the cmd file from:

`@echo off%~dp0\build\ndk-build.cmd %*`
to
`@echo off`
THAT works for me

freebumba27, 2017-03-22:
Which ndk version you are using. I am using the default ndk comes with android studio. Try to do one thing clone the project newly and try to run it then you can see it.

ZahirSoufiane, 2017-03-22:
im using external ndk [NDK Latest Stable Version (r14b)](https://developer.android.com/ndk/downloads/index.html)

freebumba27, 2017-03-22:
Let me try this give a moment.

freebumba27, 2017-03-22:
[image]
Same thing. I will suggest you to try a new laptop or Pc try to clone it and try to run in you can see the issue.

rehlma1989, 2017-03-23:
@freebumba27,
I had the same problem and could solve it, without edit the ndk-build.cmd, by **removing mips** from the build.

solution:
https://github.com/saki4510t/UVCCamera/issues/133

>libuvccamera/src/main/jni/Application.mk
 APP_ABI := armeabi armeabi-v7a x86 mips
delete mips

saki4510t, 2017-03-24:
Hi,
I update repository to improve this issue.
The issue came from that Clang(LLVM) in NDK(at least r14) does not support MSA (Mips SIMD Architecture) correctly.
If the compiler is Clang, MSA  is disabled in `Android.mk` now.

If you still need MSA, try to un-comment `#NDK_TOOLCHAIN_VERSION := 4.9` in `Application.mk` and build libraries with GCC. Although I don't recommend it because supporting GCC on NDK is already deprecated and GCC will be removed soon.

closed 2017-03-24 by saki4510t