saki4510t/UVCCamera #152: App Crash on UVC Setup

sujeeth16, 2017-04-19:
Hi,

Am trying to work with UVC Camera and Samsung S7 Android Device. Trying to test usbCameraTest app, after I click camera icon on UI, the app completely crashes with below log. I see uvc_scan_streaming is called like, 

scan_ret = uvc_scan_streaming(dev, info, block[i]);

However method signature is uvc_error_t uvc_scan_streaming(uvc_device_t *dev, uvc_device_info_t *info, int interface_idx) 

How come block being assigned to interface_idx ?

Need your help on this.

Thanks

04-18 11:45:48.324 5756-5795/? I/libusb/core: [5795*core.c:2085:libusb_init2]:libusb v1.0.19.10903
04-18 11:45:48.324 5756-5795/? I/libusb/usbfs: [5795*android_usbfs.c:504:op_init2]:start up hotplug event handler
04-18 11:45:48.324 5756-5795/? I/libusb/usbfs: [5795*android_usbfs.c:512:op_init2]:call android_scan_devices
04-18 11:45:48.324 5756-5795/? I/libusb/usbfs: [5795*android_usbfs.c:667:android_scan_devices]:__ANDROID__ do nothing
04-18 11:45:48.325 5756-5795/? A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0xd in tid 5795 (BaseActivity)
                                       
                                       [ 04-18 11:45:48.326   598:  598 W/         ]
                                       debuggerd: handling request: pid=5756 uid=10209 gid=10209 tid=5795
04-18 11:45:48.331 1549-4071/? D/InputMethodManagerService: windowGainedFocus mCurrentFocusedUserId - 0 and mSecureKeypadEnabled-false
04-18 11:45:48.393 890-1177/? I/APM_AudioPolicyManager: stopOutput() output 21, stream 1, session 169
04-18 11:45:48.418 6314-6314/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
04-18 11:45:48.419 6314-6314/? A/DEBUG: Build fingerprint: 'samsung/poseidonlteuc/poseidonlteatt:7.0/NRD90M/G891AUCU2BQB2:user/release-keys'
04-18 11:45:48.420 6314-6314/? A/DEBUG: Revision: '4'
04-18 11:45:48.420 6314-6314/? A/DEBUG: ABI: 'arm'
04-18 11:45:48.420 6314-6314/? A/DEBUG: pid: 5756, tid: 5795, name: BaseActivity  >>> com.serenegiant.usbcameratest <<<
04-18 11:45:48.420 6314-6314/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xd
04-18 11:45:48.420 6314-6314/? A/DEBUG:     r0 c7de05b0  r1 c7dfb2e0  r2 00000001  r3 00000001
04-18 11:45:48.420 6314-6314/? A/DEBUG:     r4 c6705a00  r5 00000000  r6 c7de05b0  r7 c7dfb2e0
04-18 11:45:48.420 6314-6314/? A/DEBUG:     r8 c7de05b0  r9 c7e08ee8  sl 00010e00  fp e51d5d34
04-18 11:45:48.420 6314-6314/? A/DEBUG:     ip 0000010a  sp e51d5b38  lr c754eda0  pc c754ecd8  cpsr 200f0010
04-18 11:45:48.425 6314-6314/? A/DEBUG: backtrace:
04-18 11:45:48.425 6314-6314/? A/DEBUG:     #00 pc 00007cd8  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libuvc.so (uvc_scan_streaming+212)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #01 pc 00007d9c  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libuvc.so (uvc_parse_vc_header+88)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #02 pc 00008018  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libuvc.so (uvc_scan_control+364)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #03 pc 000080a0  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libuvc.so (uvc_get_device_info+64)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #04 pc 00008170  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libuvc.so (uvc_open+84)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #05 pc 00007074  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libUVCCamera.so (_ZN9UVCCamera7connectEiiiiiPKc+248)
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #06 pc 00014790  /data/app/com.serenegiant.usbcameratest-1/lib/arm/libUVCCamera.so
04-18 11:45:48.426 6314-6314/? A/DEBUG:     #07 pc 000e3a4d  /data/app/com.serenegiant.usbcameratest-1/oat/arm/base.odex (offset 0xdf000)
04-18 11:45:48.581 1549-1857/? I/WindowManager: Destroying surface Surface(name=com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity) called by com.android.server.wm.WindowStateAnimator.destroySurface:2847 com.android.server.wm.WindowStateAnimator.destroySurfaceLocked:1078 com.android.server.wm.WindowState.destroyOrSaveSurface:2536 com.android.server.wm.AppWindowToken.destroySurfaces:412 com.android.server.wm.AppWindowToken.destroySurfaces:376 com.android.server.wm.WindowStateAnimator.finishExit:654 com.android.server.wm.WindowStateAnimator.stepAnimationLocked:564 com.android.server.wm.WindowAnimator.updateWindowsLocked:431 
04-18 11:45:48.582 709-727/? I/SurfaceFlinger: id=1039 Removed NainActivit (3/4)
04-18 11:45:48.584 1549-1857/? D/LockSettingsStorage: mSKTLockState = 2
04-18 11:45:48.588 709-12728/? I/SurfaceFlinger: id=1039 Removed NainActivit (-2/4)

zhou129311, 2017-08-15:
01-21 17:22:25.040 229-229/? A/DEBUG: backtrace:
01-21 17:22:25.040 229-229/? A/DEBUG:     #00 pc 000087a4  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libuvc.so (uvc_scan_streaming+28)
01-21 17:22:25.040 229-229/? A/DEBUG:     #01 pc 00008538  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libuvc.so (uvc_parse_vc+184)
01-21 17:22:25.041 229-229/? A/DEBUG:     #02 pc 00007fbc  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libuvc.so (uvc_scan_control+392)
01-21 17:22:25.041 229-229/? A/DEBUG:     #03 pc 000079b4  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libuvc.so (uvc_open+296)
01-21 17:22:25.041 229-229/? A/DEBUG:     #04 pc 00007cb8  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libUVCCamera.so (_ZN9UVCCamera7connectEiiiiiPKc+192)
01-21 17:22:25.041 229-229/? A/DEBUG:     #05 pc 000163f4  /data/app/com.serenegiant.usbcameratest8-1/lib/arm/libUVCCamera.so
01-21 17:22:25.041 229-229/? A/DEBUG:     #06 pc 0047ad45  /data/app/com.serenegiant.usbcameratest8-1/oat/arm/base.odex (offset 0x314000) (int com.serenegiant.usb.UVCCamera.nativeConnect(long, int, int, int, int, int, java.lang.String)+160)

mipsel-linux-android-addr2line.exe -e libuvc.so 000087a4 00008538 00007fbc 000079b4
UVCCamera\libuvccamera\src\main/jni/libuvc/android/jni/../../src/device.c:1201
UVCCamera\libuvccamera\src\main/jni/libuvc/android/jni/../../src/device.c:1004
UVCCamera\libuvccamera\src\main/jni/libuvc/android/jni/../../src/device.c:958
UVCCamera\libuvccamera\src\main/jni/libuvc/android/jni/../../src/device.c:391

device.c:1201

1188 uvc_error_t uvc_scan_streaming(uvc_device_t *dev, uvc_device_info_t *info,
1189		int interface_idx) {
1190	const struct libusb_interface_descriptor *if_desc;
1191	const unsigned char *buffer;
1192	size_t buffer_left, block_size;
1193	uvc_error_t ret, parse_ret;
1194	uvc_streaming_interface_t *stream_if;
1195
1196	UVC_ENTER();
1197
1198	ret = UVC_SUCCESS;
1199
1200	if_desc = &(info->config->interface[interface_idx].altsetting[0]);
1201	buffer = if_desc->extra; //error？？
1203	buffer_left = if_desc->extra_length;
1204	// XXX some device have it's format descriptions after the endpoint descriptor
1205	if (UNLIKELY(!buffer || !buffer_left)) {
1206		if (if_desc->bNumEndpoints && if_desc->endpoint) {
1207			// try to use extra data in endpoint[0]
1208			buffer = if_desc->endpoint[0].extra;
1209			buffer_left = if_desc->endpoint[0].extra_length;
1210		}
1211	}
1212	stream_if = calloc(1, sizeof(*stream_if));
1213	stream_if->parent = info;
1214	stream_if->bInterfaceNumber = if_desc->bInterfaceNumber;
1215	DL_APPEND(info->stream_ifs, stream_if);
1216
1217	if (LIKELY(buffer_left >= 3)) {
1218		while (buffer_left >= 3) {
1219			block_size = buffer[0];
1220//			MARK("bDescriptorType=0x%02x", buffer[1]);
1221			parse_ret = uvc_parse_vs(dev, info, stream_if, buffer, block_size);
1222
1223			if (parse_ret != UVC_SUCCESS) {
1224				ret = parse_ret;
1225				break;
1226			}
1227
1228			buffer_left -= block_size;
1229			buffer += block_size;
1230		}
1231	} else {
1232		LOGW("This VideoStreaming interface has no extra data");
1233	}
1234
1235	UVC_EXIT(ret);
1236	return ret;
1237}

saurabh111091, 2018-02-26:
Hi @sujeeth16 . were you able to get around this problem ? Will be so glad if you could help here. Thanks

saurabh111091, 2018-02-26:
Found the solution here: https://github.com/saki4510t/UVCCamera/issues/181