saki4510t/UVCCamera #156: Hardcoding the UsbDevice address instead of through dialog

ravitejachivukula, 2017-04-28:
Hi @saki4510t ,

I am relatively new to Android programming (about 2 months). I am working on an application in which I have a USB camera always connected to Android device.

I want to write a code that directly connects to the USB camera and streams the frames, instead of opening a popup asking the user to choose the USB camera. (This is very important for my requirement). To simplify my problem, there is only one USB camera connected, and it is connected always. I also want to acquire the frame data in the IFrameCallback method and pump the bitmap into my image processing code, running in another thread.

I have a few questions in this regard:

- Do you suggest, I edit the 'usbCameraTest' example code (or any other example code)? Or should I just use the UVCCamera & USBMonitor libraries and write the Main code from scratch?

- If you suggest that I edit the 'usbCameraTest' example code, can you please let me know where I need to modify the code? I tried figuring it out, but I am a little confused about where I need to make the changes so that instead of displaying a dialog, the code directly takes a hardcoded UsbDevice address...

- If you suggest that I write the code from scratch, it is my understanding that only UVCCamera and USBMonitor classes will be sufficient for my purpose. Is my understanding correct?

Thanks a lot in advance.
Raviteja

ravitejachivukula, 2017-04-29:
Hi @saki4510t ,

I did try to arrive at, what I think is the bare minimum version of the code for my requirement. I have included it below:

    protected void onCreate(Bundle savedInstanceState) {

        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        mUsbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        HashMap<String, UsbDevice> deviceList = mUsbManager.getDeviceList();

        mUsbDevicename = (String)deviceList.keySet().toArray()[0];
        mUsbDevice = deviceList.get(mUsbDevicename);
        //UsbDevice mUsbDevice = deviceList.get("/dev/bus/usb/001/002");
        Log.d("Execution State","USB Device = "+mUsbDevice);

        USBMonitor mUSBMonitor = new USBMonitor(this,mOnDeviceConnectListener);
        mUSBMonitor.register();

        String ACTION_USB_PERMISSION = "com.android.example.USB_PERMISSION";
        PendingIntent mPermissionIntent = PendingIntent.getBroadcast(this,0,new Intent(ACTION_USB_PERMISSION),0);
        mUsbManager.requestPermission(mUsbDevice,mPermissionIntent);

        mUsbControlBlock = mUSBMonitor.openDevice(mUsbDevice);

        mUVCCamera = new UVCCamera();
        mUVCCamera.open(mUsbControlBlock);

        mUVCCamera.setFrameCallback(mIFrameCallback,UVCCamera.PIXEL_FORMAT_RGB565);

    }

    private final IFrameCallback mIFrameCallback = new IFrameCallback() {
        @Override
        public void onFrame(final ByteBuffer frame) {
            frame.clear();

            Log.d("Execution State","mIFrameCallback - Frame Received!!");
        }
    };

    private final USBMonitor.OnDeviceConnectListener mOnDeviceConnectListener = new USBMonitor.OnDeviceConnectListener() {
        @Override
        public void onAttach(final UsbDevice device) {
        }

        @Override
        public void onConnect(final UsbDevice device, final USBMonitor.UsbControlBlock ctrlBlock, final boolean createNew) {
        }
        @Override
        public void onDisconnect(final UsbDevice device, final USBMonitor.UsbControlBlock ctrlBlock) {
        }

        @Override
        public void onDettach(final UsbDevice device) {
        }

        @Override
        public void onCancel(final UsbDevice device) {
        }

        };
}

I expect that this code will wait for a frame from the USB camera and execute the IFrameCallback method on the receipt of the frame (for now the IFrameCallback method just logs onto logcat). But it is not working as I expect. The code is getting compiled without any errors, the APK gets deployed, and there are no Runtime errors. But IFrameCallback is not getting executed!

I am pretty sure, it has to do with my lack of understanding of how the code works. Please let me know where you think I am going wrong...

Thanks & Regards
Raviteja

saki4510t, 2017-04-30:
Hi,

You need to handle permission correctly in AndroidManifest.xml and/or
OnDeviceConnectListener.
And as I wrote in previous issue on UVCCamera repository, IFrameCallback is
called after start preview only(between start and stop previewing).
So you need to supply Surface/SurfaceTexture as preview display/screen and
start preview.

with best regards,
saki