saki4510t/UVCCamera #164: Error 0xfffffc0e

enight0000, 2017-05-16:
I am using usbcameratest3 for a record, when I changed the preview size to 2560x720, this error occured, what's the problem and how to solve it? Thank you.

com.serenegiant.usbcameratest3 E/AndroidRuntime: FATAL EXCEPTION: CameraThread
Process: com.serenegiant.usbcameratest3, PID: 3091
android.media.MediaCodec$CodecException: Error 0xfffffc0e
at android.media.MediaCodec.native_configure(Native Method)
at android.media.MediaCodec.configure(MediaCodec.java:580)                                                                                  at com.serenegiant.encoder.MediaVideoEncoder.prepare(MediaVideoEncoder.java:106)                                                                                  at com.serenegiant.encoder.MediaMuxerWrapper.prepare(MediaMuxerWrapper.java:76)                                                                                  at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.handleStartRecording (AbstractUVCCameraHandler.java:565)
at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler.handleMessage (AbstractUVCCameraHandler.java:315)
at com.serenegiant.usbcameracommon.UVCCameraHandler.handleMessage(UVCCameraHandler.java:31)
at android.os.Handler.dispatchMessage(Handler.java:102)                                                                                  at android.os.Looper.loop(Looper.java:135)
at com.serenegiant.usbcameracommon.AbstractUVCCameraHandler$CameraThread.run (AbstractUVCCameraHandler.java:747)

kingjie0210, 2017-05-17:
How did you change the preview size? you change it in the code or manually selecting the resolution 
on the interface?

enight0000, 2017-05-17:
I changed the variable in MainActivity.class, line 66
private static final int PREVIEW_WIDTH = 2560;

And it's ok to preview the video, but when I press the record button, error occured!

ldima, 2017-07-25:
If you set not supported (High) parameters for recording you will this error.
it can be  resolution, fps, profile, level and other.
You can catch this error and try recording on lower perameters.

closed 2017-10-16 by enight0000