saki4510t/UVCCamera #167: Camera Preview Sizes  issue : no camera preview  is showing if change preview height width

aditayDevelopment, 2017-05-18:
hello 

I am working on Project where need to use usb camera in kiosk. when i try to attached your library in project then its working fine when i use web camera with default preview height, width. but when i sent apk to client he checked on kiosk , and camera detected but not any preview displaying, i  am just guessing  is it a problem  of supported camera preview height width in kiosk i don't know the supported screen sizes of kiosk camera . is i am right? how can i solve this problem.

RogerHardiman, 2017-06-02:
I had a similar problem. My Wide Angle USB camera only has 640x400 resolution (and not 640x480).

After ```camera.open()``` I call the ```camera.getSupportedSizeList()``` function to get a list of valid sizes. I then pick the best size from the list and use the valid width/height from the list when calling ```camera.setPreviewSize()```

That gives me a single APK that works on old UVC cameras and on Wide Angle cameras.

Hint - for a quick check of camera sizes you can run this command on Linux.
```v4l2-ctl --list-formats-ext ```

PR #175 has an example of this.

aditayDevelopment, 2017-06-02:
so your app working?

RogerHardiman, 2017-06-02:
Have you based your code on one of the existing Test programs?

aditayDevelopment, 2017-06-02:
yes

RogerHardiman, 2017-06-02:
which one

aditayDevelopment, 2017-06-02:
??

RogerHardiman, 2017-06-02:
which test program have you used. There are 9 sample programs (numbered Test0 to Test8)

aditayDevelopment, 2017-06-02:
test 6

RogerHardiman, 2017-06-02:
My changes were in Test0. I'm not sure exactly where to probe for the size in Test6

aditayDevelopment, 2017-06-02:
please check my log

E/TAG     ( 1337): Crash report
I/com.nullwire.trace.ExceptionsHandler( 1337): Registering default exceptions handler: http://pratham.cwsdev3.biz/app/server.php
I/com.nullwire.trace.ExceptionsHandler( 1337): Registering default exceptions handler
I/com.nullwire.trace.ExceptionsHandler( 1337): TRACE_VERSION: 0.3.0
D/com.nullwire.trace.ExceptionsHandler( 1337): APP_VERSION: 1.0
D/com.nullwire.trace.ExceptionsHandler( 1337): APP_PACKAGE: com.selfieapp
D/com.nullwire.trace.ExceptionsHandler( 1337): FILES_PATH: /data/data/com.selfieapp/files
D/com.nullwire.trace.ExceptionsHandler( 1337): URL: http://pratham.cwsdev3.biz/app/server.php
W/GooglePlayServicesUtil( 1337): Google Play Store is missing.
W/FA      ( 1337): Service invalid
D/com.nullwire.trace.ExceptionsHandler( 1337): Looking for exceptions in: /data/data/com.selfieapp/files
D/com.nullwire.trace.ExceptionsHandler( 1337): current handler class=com.android.internal.os.RuntimeInit$UncaughtHandler
I/dalvikvm( 1337): Could not find method android.view.Window$Callback.onProvideKeyboardShortcuts, referenced from method android.support.v7.view.WindowCallbackWrapper.onProvideKeyboardShortcuts
W/dalvikvm( 1337): VFY: unable to resolve interface method 19959: Landroid/view/Window$Callback;.onProvideKeyboardShortcuts (Ljava/util/List;Landroid/view/Menu;I)V
D/dalvikvm( 1337): VFY: replacing opcode 0x72 at 0x0002
W/dalvikvm( 1337): VFY: unable to find class referenced in signature (Landroid/view/SearchEvent;)
I/dalvikvm( 1337): Could not find method android.view.Window$Callback.onSearchRequested, referenced from method android.support.v7.view.WindowCallbackWrapper.onSearchRequested
W/dalvikvm( 1337): VFY: unable to resolve interface method 19961: Landroid/view/Window$Callback;.onSearchRequested (Landroid/view/SearchEvent;)Z
D/dalvikvm( 1337): VFY: replacing opcode 0x72 at 0x0002
I/dalvikvm( 1337): Could not find method android.view.Window$Callback.onWindowStartingActionMode, referenced from method android.support.v7.view.WindowCallbackWrapper.onWindowStartingActionMode
W/dalvikvm( 1337): VFY: unable to resolve interface method 19965: Landroid/view/Window$Callback;.onWindowStartingActionMode (Landroid/view/ActionMode$Callback;I)Landroid/view/ActionMode;
D/dalvikvm( 1337): VFY: replacing opcode 0x72 at 0x0002
I/dalvikvm( 1337): Could not find method android.content.res.TypedArray.getChangingConfigurations, referenced from method android.support.v7.widget.TintTypedArray.getChangingConfigurations
W/dalvikvm( 1337): VFY: unable to resolve virtual method 720: Landroid/content/res/TypedArray;.getChangingConfigurations ()I
D/dalvikvm( 1337): VFY: replacing opcode 0x6e at 0x0002
I/dalvikvm( 1337): Could not find method android.content.res.TypedArray.getType, referenced from method android.support.v7.widget.TintTypedArray.getType
W/dalvikvm( 1337): VFY: unable to resolve virtual method 742: Landroid/content/res/TypedArray;.getType (I)I
D/dalvikvm( 1337): VFY: replacing opcode 0x6e at 0x0008
D/dalvikvm( 1337): GC_CONCURRENT freed 361K, 13% free 3058K/3488K, paused 7ms+3ms, total 27ms
I/dalvikvm( 1337): Could not find method android.content.res.Resources.getDrawable, referenced from method android.support.v7.widget.ResourcesWrapper.getDrawable
W/dalvikvm( 1337): VFY: unable to resolve virtual method 683: Landroid/content/res/Resources;.getDrawable (ILandroid/content/res/Resources$Theme;)Landroid/graphics/drawable/Drawable;
D/dalvikvm( 1337): VFY: replacing opcode 0x6e at 0x0002
I/dalvikvm( 1337): Could not find method android.content.res.Resources.getDrawableForDensity, referenced from method android.support.v7.widget.ResourcesWrapper.getDrawableForDensity
W/dalvikvm( 1337): VFY: unable to resolve virtual method 685: Landroid/content/res/Resources;.getDrawableForDensity (IILandroid/content/res/Resources$Theme;)Landroid/graphics/drawable/Drawable;
D/dalvikvm( 1337): VFY: replacing opcode 0x6e at 0x0002
E/dalvikvm( 1337): Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method android.support.v7.widget.AppCompatImageHelper.hasOverlappingRendering
W/dalvikvm( 1337): VFY: unable to resolve instanceof 202 (Landroid/graphics/drawable/RippleDrawable;) in Landroid/support/v7/widget/AppCompatImageHelper;
D/dalvikvm( 1337): VFY: replacing opcode 0x20 at 0x000c
I/dalvikvm( 1337): Could not find method android.content.Context.getColorStateList, referenced from method android.support.v7.content.res.AppCompatResources.getColorStateList
W/dalvikvm( 1337): VFY: unable to resolve virtual method 470: Landroid/content/Context;.getColorStateList (I)Landroid/content/res/ColorStateList;
D/dalvikvm( 1337): VFY: replacing opcode 0x6e at 0x0006
D/dalvikvm( 1337): GC_FOR_ALLOC freed 90K, 10% free 3145K/3488K, paused 13ms, total 13ms
I/dalvikvm-heap( 1337): Grow heap (frag case) to 3.403MB for 281896-byte allocation
D/dalvikvm( 1337): GC_FOR_ALLOC freed 1K, 10% free 3420K/3764K, paused 12ms, total 12ms
D/dalvikvm( 1337): GC_FOR_ALLOC freed 13K, 9% free 3445K/3764K, paused 12ms, total 12ms
I/ActivityManager(  393): START u0 {cmp=com.selfieapp/.LoginSceenActivity} from pid 1337
V/FirebaseCrash( 1337): Firebase Crash Reporting is disabled.
W/Settings( 1337): Setting airplane_mode_on has moved from android.provider.Settings.System to android.provider.Settings.Global, returning read-only value.
E/TAG     ( 1337): 0
[... cut ...]
I/DEBUG   (   90):     000000c0 ffffffff ffffffff ffffffff ffffffff  
I/DEBUG   (   90):     000000d0 ffffffff ffffffff ffffffff ffffffff  
I/DEBUG   (   90):     000000e0 ffffffff ffffffff ffffffff ffffffff  
I/DEBUG   (   90):     000000f0 ffffffff ffffffff ffffffff ffffffff  
I/DEBUG   (   90): 
I/DEBUG   (   90): memory map around fault addr 71152000:
I/DEBUG   (   90):     70151000-71152000 rw- 
I/DEBUG   (   90):     (no map for address)
I/DEBUG   (   90):     71414000-71bfd000 rw- /dev/ump