saki4510t/UVCCamera #170: UVCService 一直显示警告信息

akingyin1987, 2017-05-23:
05-23 15:41:51.569 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.569 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.569 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.599 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.599 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.599 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.609 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.609 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.609 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.609 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.629 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.639 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.669 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.699 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.709 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.739 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.769 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.799 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.809 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-ES20: <core_glFlush:52>: GL_OUT_OF_MEMORY
05-23 15:41:51.839 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.839 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.839 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.839 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
[... cut ...]
05-23 15:41:51.869 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/GLConsumer: [unnamed-31522-1] bindTextureImage: clearing GL error: 0x505
05-23 15:41:51.869 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0x0, handle=0x0
05-23 15:41:51.869 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGL: <qeglDrvAPI_eglSwapBuffers:3712>: EGL_BAD_SURFACE
05-23 15:41:51.869 31522-32607/com.zlcdgroup.client_mrmsei:uvcservice W/Adreno-EGLSUB: <DequeueBuffer:721>: dequeue native buffer fail: No such device, buffer=0

pocketfpv, 2017-05-23:
是不是手机太老旧了。不支持openGL es 2.0 ？
你这是什么手机？

akingyin1987, 2017-05-24:
三星J3，我了解了Surface没有移除，UVCService 方法 checkReleaseService BUG  现在已经解决了