saki4510t/UVCCamera #173: Fatal signal 11 (SIGSEGV) and signal 6 (SIGABRT)

rehlma1989, 2017-05-26:
For some Android devices, I get a native error.

I've tried it for 4 devices. 2 (Android 6.0.1) are working and 2 (Android 6.0 and 5.1.1) are not working.
APP_ABI := armeabi armeabi-v7a x86

Here a few different reports.

**logcat with APP_OPTIM := release**
```
05-27 05:07:42.814 14116 14189 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0xdde00000 in tid 14189 (Thread-408)
05-27 05:07:42.866  3321  3321 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
05-27 05:07:42.866  3321  3321 F DEBUG   : Build fingerprint: 'Letv/U4/U4:6.0/V3301RCN02C059055B03142S/20170314:user/release-keys'
05-27 05:07:42.866  3321  3321 F DEBUG   : Revision: '0'
05-27 05:07:42.866  3321  3321 F DEBUG   : ABI: 'arm'
05-27 05:07:42.867  3321  3321 F DEBUG   : pid: 14116, tid: 14189, name: Thread-408  >>> com.serenegiant.usbcameratest0 <<<
05-27 05:07:42.867  3321  3321 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xdde00000
05-27 05:07:42.898  3321  3321 F DEBUG   :     r0 dddffff5  r1 de46b980  r2 00000100  r3 039946b5
05-27 05:07:42.899  3321  3321 F DEBUG   :     r4 00000280  r5 00000000  r6 de46b801  r7 00000280
05-27 05:07:42.899  3321  3321 F DEBUG   :     r8 ef652bc0  r9 ef622f10  sl de46b800  fp de43f618
05-27 05:07:42.899  3321  3321 F DEBUG   :     ip f73ec5dc  sp de43f5d0  lr debd4fcc  pc f738ec8e  cpsr 20070030
05-27 05:07:42.907  3321  3321 F DEBUG   : 
05-27 05:07:42.907  3321  3321 F DEBUG   : backtrace:
05-27 05:07:42.907  3321  3321 F DEBUG   :     #00 pc 00017c8e  /system/lib/libc.so (__memcpy_base+193)
05-27 05:07:42.907  3321  3321 F DEBUG   :     #01 pc 00014fc8  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libuvc.so
05-27 05:07:42.907  3321  3321 F DEBUG   :     #02 pc 0000c314  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+320)
05-27 05:07:42.908  3321  3321 F DEBUG   :     #03 pc 000126a8  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so
05-27 05:07:42.908  3321  3321 F DEBUG   :     #04 pc 0000d200  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so
05-27 05:07:42.908  3321  3321 F DEBUG   :     #05 pc 0000cb60  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968)
05-27 05:07:42.908  3321  3321 F DEBUG   :     #06 pc 0000d3cc  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (libusb_handle_events+52)
05-27 05:07:42.908  3321  3321 F DEBUG   :     #07 pc 00012c20  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libuvc.so (_uvc_handle_events+136)
05-27 05:07:42.908  3321  3321 F DEBUG   :     #08 pc 0003fd13  /system/lib/libc.so (_ZL15__pthread_startPv+30)
05-27 05:07:42.909  3321  3321 F DEBUG   :     #09 pc 0001a345  /system/lib/libc.so (__start_thread+6)
05-27 05:07:43.230  3644 14216 W ActivityManager:   Force finishing activity com.serenegiant.usbcameratest0/.MainActivity
```

**ndk-stack with APP_OPTIM := release**
```
********** Crash dump: **********
Build fingerprint: 'Letv/U4/U4:6.0/V3301RCN02C059055B03142S/20170314:user/release-keys'
pid: 14116, tid: 14189, name: Thread-408  >>> com.serenegiant.usbcameratest0 <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xdde00000
Stack frame 05-27 05:07:42.907  3321  3321 F DEBUG   :     #00 pc 00017c8e  /system/lib/libc.so (__memcpy_base+193)
Stack frame 05-27 05:07:42.907  3321  3321 F DEBUG   :     #01 pc 00014fc8  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libuvc.so: Routine uvc_stream_open_ctrl at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libuvc/android/jni/../../src/stream.c:1356
Stack frame 05-27 05:07:42.907  3321  3321 F DEBUG   :     #02 pc 0000c314  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (usbi_handle_transfer_completion+320): Routine usbi_parse_descriptor at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/descriptor.c:97
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #03 pc 000126a8  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so: Routine op_cancel_transfer at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/os/android_usbfs.c:2545
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #04 pc 0000d200  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so: Routine list_del at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/libusbi.h:129
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #05 pc 0000cb60  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (libusb_handle_events_timeout_completed+968): Routine libusb_hotplug_register_callback at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/hotplug.c:249
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #06 pc 0000d3cc  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libusb100.so (libusb_handle_events+52): Routine calculate_timeout at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:1218
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #07 pc 00012c20  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libuvc.so (_uvc_handle_events+136): Routine uvc_mjpeg2bgr at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libuvc/android/jni/../../src/frame-mjpeg.c:293
Stack frame 05-27 05:07:42.908  3321  3321 F DEBUG   :     #08 pc 0003fd13  /system/lib/libc.so (_ZL15__pthread_startPv+30)
Stack frame 05-27 05:07:42.909  3321  3321 F DEBUG   :     #09 pc 0001a345  /system/lib/libc.so (__start_thread+6)
```

**logcat with APP_OPTIM := debug**
```
05-27 05:48:13.873 21228 21346 F libc    : jni/libuvc/android/jni/../../src/stream.c:932: _uvc_process_payload_iso: assertion "strmh->got_bytes + odd_bytes < strmh->size_buf" failed
05-27 05:48:13.873 21228 21346 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 21346 (Thread-453)
05-27 05:48:13.925  3321  3321 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
05-27 05:48:13.925  3321  3321 F DEBUG   : Build fingerprint: 'Letv/U4/U4:6.0/V3301RCN02C059055B03142S/20170314:user/release-keys'
05-27 05:48:13.925  3321  3321 F DEBUG   : Revision: '0'
05-27 05:48:13.925  3321  3321 F DEBUG   : ABI: 'arm'
05-27 05:48:13.926  3321  3321 F DEBUG   : pid: 21228, tid: 21346, name: Thread-453  >>> com.serenegiant.usbcameratest0 <<<
05-27 05:48:13.926  3321  3321 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
05-27 05:48:13.966  3321  3321 F DEBUG   : Abort message: 'jni/libuvc/android/jni/../../src/stream.c:932: _uvc_process_payload_iso: assertion "strmh->got_bytes + odd_bytes < strmh->ize_buf" failed'
05-27 05:48:13.967  3321  3321 F DEBUG   :     r0 00000000  r1 00005362  r2 00000006  r3 de548978
05-27 05:48:13.967  3321  3321 F DEBUG   :     r4 de548980  r5 de548930  r6 0000000b  r7 0000010c
05-27 05:48:13.967  3321  3321 F DEBUG   :     r8 e8fe9b20  r9 e8fff180  sl df957dcc  fp de5485d0
05-27 05:48:13.967  3321  3321 F DEBUG   :     ip 00000006  sp de5484f0  lr f73b7415  pc f73b9050  cpsr 40070010
05-27 05:48:13.980  3321  3321 F DEBUG   : 
05-27 05:48:13.980  3321  3321 F DEBUG   : backtrace:
[... 19 lines cut ...]

**ndk-stack with APP_OPTIM := debug**
```
********** Crash dump: **********
Build fingerprint: 'Letv/U4/U4:6.0/V3301RCN02C059055B03142S/20170314:user/release-keys'
pid: 21228, tid: 21346, name: Thread-453  >>> com.serenegiant.usbcameratest0 <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #00 pc 00042050  /system/lib/libc.so (tgkill+12)
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #01 pc 00040411  /system/lib/libc.so (pthread_kill+32)
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #02 pc 0001cb0f  /system/lib/libc.so (raise+10)
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #03 pc 00019cb1  /system/lib/libc.so (__libc_android_abort+34)
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #04 pc 00017a64  /system/lib/libc.so (abort+4)
Stack frame 05-27 05:48:13.981  3321  3321 F DEBUG   :     #05 pc 0001b723  /system/lib/libc.so (__libc_fatal+16)
[... cut ...]
Stack frame 05-27 05:48:13.983  3321  3321 F DEBUG   :     #14 pc 0001b1f8  /data/app/com.serenegiant.usbcameratest0-1/lib/arm/libuvc.so (_uvc_handle_events+256): Routine _uvc_handle_events at C:\Users\Admin\Downloads\UVCCamera-master\UVCCamera-master\libuvccamera\src\main/jni/libuvc/android/jni/../../src/init.c:107
Stack frame 05-27 05:48:13.983  3321  3321 F DEBUG   :     #15 pc 0003fd13  /system/lib/libc.so (_ZL15__pthread_startPv+30)
Stack frame 05-27 05:48:13.983  3321  3321 F DEBUG   :     #16 pc 0001a345  /system/lib/libc.so (__start_thread+6)
```

saurabh111091, 2018-02-26:
@rehlma1989 This solved it for me : https://github.com/saki4510t/UVCCamera/issues/181

ysyyork, 2018-07-10:
@rehlma1989 Hi, I also get same error but I can't print the stack trace. What should I do to get that? Now I got same error as the first example you posted, but it only prints out something like `07-10 13:12:31.193  1598  1669 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x7d30400000 in tid 1669 (Thread-20)`

MatthewDevelop, 2019-12-03:
@ysyyork Hi,could you tell me how did you solve this problem