saki4510t/UVCCamera #174: Opening camera fails on Android 7.0

DrakkLord, 2017-05-30:
Using UVCCamera on an Android 7.0 device results in open failure and the following logcat output

> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [libusb_init2] created default context
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [libusb_init2] libusb v1.0.19.10903
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [op_init2] bulk continuation flag supported
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [op_init2] zero length packet flag supported
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [op_init2] sysfs can relate devices
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/libusb: [op_init2] sysfs has complete descriptors
> 05-30 11:54:24.535 3762-3762/aero.glass.android E/libusb: [op_init2] warning: error starting hotplug event monitor
> 05-30 11:54:24.535 3762-3762/aero.glass.android E/libusb: [sysfs_get_device_list] opendir devices failed errno=13
> 05-30 11:54:24.535 3762-3762/aero.glass.android D/UsbDeviceConnectionJNI: close
> 05-30 11:54:24.536 3762-3762/aero.glass.android E/USBCameraHandler: Failed to open camera : /dev/bus/usb/001/004

the same exact APK works fine on 6.0.1, is there a way to make it work on 7.0 if not at the moment are there any plans to make it work, without root?

RogerHardiman, 2017-06-02:
I just tried USBTest0 on a Samsung Galaxy S7 with Android 7.0. It worked fine.
Do you get the same errors with the Test Programmes on your Android 7 device?

DrakkLord, 2017-06-02:
I've just tested it on a Sony Xperia Z5 compact on Android 7.0, it worked.
It was my mistake , I was using an older version, the current master branch worked fine.

closed 2017-06-02 by DrakkLord