saki4510t/UVCCamera #178: How to resolve USB camera bandwidth issue

LCS1969, 2017-06-05:
Dear all

I have an issue (USB bandwidth error) when I installed a USB camera and run "usbCameraTest" app, the message from log file as below

_06-05 11:27:17.640 17407-18443/com.serenegiant.usbcameratest E/libuvc/stream: [18443*stream.c:1593:uvc_stream_start_bandwidth]:libusb_submit_transfer failed
06-05 11:27:17.640 17407-18443/com.serenegiant.usbcameratest E/libuvc/stream: [18443*stream.c:1606:uvc_stream_start_bandwidth]:fail
06-05 11:27:17.640 17407-18443/com.serenegiant.usbcameratest W/libUVCCamera: [18443*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)
06-05 11:28:17.660 17407-17407/com.serenegiant.usbcameratest W/libUVCCamera: [17407*UVCPreview.cpp:362:stopPreview]:UVCPreview::terminate capture thread: pthread_join failed_

Trying to debug stream.c and IO.c under libusb library and the error is actually coming from below code, LINE #1469 in IO.c file,

`r = usbi_backend->submit_transfer(itransfer);`

Any suggestion on how to resolve this?

Regards,

jp1017, 2017-06-05:
Only one UVC had this issue?

I had this issue when open two UVC the same time. just like #8

jp1017, 2017-11-14:
Try using Ndk version r14b or r13b.