saki4510t/UVCCamera #180: Multiple Cameras

jp1017, 2017-06-07:
same issue #8 
but I can't resolved, and I created this new one.

error: 
```
I/libusb/core: [8636*core.c:2085:libusb_init2]:libusb v1.0.19.10903
I/libusb/usbfs: [8636*android_usbfs.c:512:op_init2]:call android_scan_devices
I/libUVCCamera: [8670*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@MJPEG
I/libuvc/stream: [8670*stream.c:1438:uvc_stream_start_bandwidth]:isochronous transfer mode:num_altsetting=5
E/libusb/usbfs: [8670*android_usbfs.c:2421:submit_iso_transfer]:submiturb failed error -1 errno=28
E/libusb/usbfs: [8670*android_usbfs.c:2430:submit_iso_transfer]:first URB failed, easy peasy
E/libuvc/stream: [8670*stream.c:1596:uvc_stream_start_bandwidth]:libusb_submit_transfer failed, ret=-1
E/libuvc/stream: [8670*stream.c:1609:uvc_stream_start_bandwidth]:fail, ret=-1
W/libUVCCamera: [8670*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)
```

I can assume this method 
>		int r = ioctl(dpriv->fd, IOCTL_USBFS_SUBMITURB, urbs[i]);
return -1, means: "IO error", so, how to resolve it.

jp1017, 2017-06-07:
error 28 means : ENOSPC: No space left on device (POSIX.1).

jp1017, 2017-11-14:
1. 使用MJPEG格式压缩的摄像头
1. 设置分辨率 320*240或者更低

closed 2017-11-14 by jp1017

liuenbao, 2017-12-04:
hello，按照你的说的配置两个条件，还是同样的错误。有什么建议么？

jp1017, 2017-12-04:
@liuenbao 你的摄像头怎么确定是MJPEG的? 或者供电不足?

liuenbao, 2017-12-04:
@jp1017 确定是MJPEG输出的，摄像头连接在一个USB-HUB上，输出320*240的30帧。就是上面一模一样错误。