saki4510t/UVCCamera #181: NDK 15 problem

argusroot, 2017-06-13:
Had success building with NDK 15, but the application fails to open the camera or crashes with other errors.

Downgrading to NDK 14b solved the problem.

xwpcom, 2017-06-14:
yes,I also have the same problem.

06-14 16:48:12.511 21141-21178/com.serenegiant.usbcameratest4:uvcservice E/UVCCamera: [21178*UVCCamera.cpp:172:connect]:could not open camera:err=-1
06-14 16:48:12.511 21141-21178/com.serenegiant.usbcameratest4:uvcservice W/dalvikvm: threadid=17: thread exiting with uncaught exception (group=0x4169dba8)
06-14 16:48:12.514 21141-21178/com.serenegiant.usbcameratest4:uvcservice E/AndroidRuntime: FATAL EXCEPTION: CameraThread
                                                                                           Process: com.serenegiant.usbcameratest4:uvcservice, PID: 21141
                                                                                           java.lang.UnsupportedOperationException: open failed:result=-1
                                                                                               at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                                                               at com.serenegiant.service.CameraServer$CameraThread.handleOpen(CameraServer.java:379)
                                                                                               at com.serenegiant.service.CameraServer.handleMessage(CameraServer.java:262)
                                                                                               at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                                               at android.os.Looper.loop(Looper.java:136)
                                                                                               at com.serenegiant.service.CameraServer$CameraThread.run(CameraServer.java:632)

xwpcom, 2017-06-14:
@argusroot 
How to downgrading to NDK 14b?

argusroot, 2017-06-14:
@xwpcom
- Remove the current NDK from Android Studio
- Download and unzip the older version from here: _https://developer.android.com/ndk/downloads/older_releases.html_
- Change the **local.properties** file to point to the extracted folder in the line **ndk.dir=**
- Have fun :grimacing:

xwpcom, 2017-06-15:
@argusroot
Thank you very much!

ThomasCZ, 2017-06-15:
@argusroot 
Did help to you older NDK? I think, problem is caused by SELinux enforces.

argusroot, 2017-06-15:
@ThomasCZ Yes. 100% working with NDK 14b

ThomasCZ, 2017-06-15:
Ok, thanks, looks like on another problem than [mine](https://github.com/saki4510t/UVCCamera/issues/177).

MaxElkin, 2017-06-19:
@argusroot Thank you very much! You saved me from despair

RogerHardiman, 2017-06-29:
Also had to downgrade to NDK 14

jerry0864, 2017-07-12:
@argusroot  do you ever got errors like "bitmap too large to be uploaded into a texture max 4096x4096" under NDK 14?

argusroot, 2017-07-12:
@jerry0864 I just tested it with the NDK 14b and until now had no errors like yours.

xbesung, 2017-07-21:
I also had same problems with NDK15 and eventually found that clang was the reason.
Then I change the compiler to gcc by uncomment "NDK_TOOLCHAIN_VERSION := 4.9" from "libuvccamera/src/main/jni/Application.mk".
And it works now.

ysyyork, 2017-08-08:
I also came across this issue. Should I keep on using NDK 14b or is there a plan to solve this issue and work with 15?

ysyyork, 2017-08-08:
@xbesung But it shows 
"
# This is just for mips, if you really needs MSA, un-comment and build with GCC.
# Note: Supporting GCC on NDK is already deprecated and GCC will be removed from NDK soon.

"
above this line. Will uncomment this line cause later issue?

xbesung, 2017-08-09:
@ysyyork GCC was the default compiler until NDK13. I think it should be fine. And this is just a solution for NDK15. You could comment it again if clang from later versions works.

BernardJiang, 2017-08-17:
Thanks for the solution!

fredo-, 2017-08-19:
I changed the NDK version to 14b, by following @argusroot's instructions, and then I also had to uncomment the line @xbesung mentioned and I was able to connect my camera. I never tried just uncommenting the line with NDK 15, so that might work.

Eversee22, 2017-08-25:
Yes, it works

davidhubbard, 2017-09-13:
Changing to r14b also worked for me. What about adding this in libuvccamera/src/main/jni/UVCCamera/UVCCamera.cpp, after `LOGE("could not find camera:err=%d.", result);`

    #if __ANDROID_API__ > 14
      LOGE("Try using Android NDK r14b https://github.com/saki4510t/UVCCamera/issues/181");
    #endif

I found a reference for __ANDROID_API__ here: https://sourceforge.net/p/predef/mailman/message/27397690/

jiangdongguo, 2017-09-27:
com.serenegiant.usb.UVCCamera.open(UVCCamera.java:217)
change to ndk14，it works! Thanks!

By the way , my question is：
com.serenegiant.usbcameratest0.MainActivity$2$1.run(MainActivity.java:149)
android.os.Handler.handleCallback(Handler.java:751)
android.os.Handler.dispatchMessage(Handler.java:95)
android.os.Looper.loop(Looper.java:154)
android.os.HandlerThread.run(HandlerThread.java:61)
open failed:result=-1

jiangdongguo, 2017-09-27:
change to ndk14，it works! Thanks!

By the way , my question is：
com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
com.serenegiant.usbcameratest0.MainActivity$2$1.run(MainActivity.java:149)
android.os.Handler.handleCallback(Handler.java:751)
android.os.Handler.dispatchMessage(Handler.java:95)
android.os.Looper.loop(Looper.java:154)
android.os.HandlerThread.run(HandlerThread.java:61)
open failed:result=-1

amani-mohammad, 2017-09-27:
I tried ndk14 on a testing app, and it works.
But when I tried it on my original app, it doesn't work at all. 
It crashes when I try to create UVCCamera object: 
`UVCCamera camera = new UVCCamera();`

evgenek11, 2017-10-09:
(Hopefully) I've managed to track down the root case of this issue.
After gcc->clang transition (NDK r14 -> NDK r15), this line isn't working as expected in

UVCCamera/blob/master/libuvccamera/src/main/jni/libusb/libusb/descriptor.c (line 299):
[https://github.com/saki4510t/UVCCamera/blob/master/libuvccamera/src/main/jni/libusb/libusb/descriptor.c](url)
			usbi_parse_descriptor(buffer, "bb", &header, 0);

This func has (int) return value, which is ignored.

Changing to 
                      int len = usbi_parse_descriptor(buffer, "bb", &header, 0);

fixes the problem.

I have no scientific explanation why this happens (ignoring return value in C doesn't look like a crime))).
It is possible that compiling using clang with optimization level -O2 causes the mess with pointer adress manipulation (in usbi_parse_descriptor).

There are other places in that file, where that function is being called (ignoring ret value as well), but this is the only fix that was required.
After applying this code modification, the default (r15) NDK can be used, no need to change _local.properties_ file.

I'd be happy to know if this fix works for others.

RogerHardiman, 2017-10-09:
@evgenek11 
You are a hero.
Thank you for tracking down the cause and sharing the solution with us.

I'd been using NDK 15 with the flag to force GCC compilation (instead of clang) but never had time to try and follow the code.

I hope @saki4510t will be able to apply this fix to the project.

evgenek11, 2017-10-10:
@RogerHardiman , thanks!

Found even better solution: reducing the optimization level from -O3 (originally) to -O1- solves the problem with no code modification.
Line 59 of _libusb.mk_ must be changed.

UVCCamera/libuvccamera/src/main/jni/libusb/android/jni/libusb.mk (line 59):

LOCAL_CFLAGS += -O3 -fstrict-aliasing -fprefetch-loop-arrays

to 

LOCAL_CFLAGS += -O1 -fstrict-aliasing -fprefetch-loop-arrays

More details about optimization issues - here:
https://stackoverflow.com/questions/2722302/can-compiler-optimization-introduce-bugs

hpsaturn, 2017-10-12:
@evgenek11 your solution, do camera led turn on come back, but I had the same problem with lastest NDK version, but is strangly, before it, the led turn off all time. The optimization flags fix not work for me. The NDK downgrade not is a solution for me because my project fails with old NDK version. 

```java
app.android.usbcamera I/libusb/core: [19208*core.c:2085:libusb_init2]:libusb v1.0.19.10903
app.android.usbcamera I/libusb/usbfs: [19208*android_usbfs.c:512:op_init2]:call android_scan_devices
app.android.usbcamera E/libusb/usbfs: [19208*android_usbfs.c:1776:release_interface]:release interface failed, error -1 errno 22
app.android.usbcamera E/UVCCamera: [19208*UVCCamera.cpp:172:connect]:could not open camera:err=-99
app.android.usbcamera W/dalvikvm: threadid=25: thread exiting with uncaught exception (group=0x41b59ba8)
app.android.usbcamera D/SensorManager: aidl getSensorManager:android.hardware.ISensorManager$Stub$Proxy@41e9d738
app.android.usbcamera D/dalvikvm: GC_FOR_ALLOC freed 492K, 12% free 4622K/5204K, paused 14ms, total 14ms
app.android.usbcamera D/dalvikvm: GC_FOR_ALLOC freed 713K, 16% free 4541K/5348K, paused 13ms, total 13ms
app.android.usbcamera E/AndroidRuntime: FATAL EXCEPTION: USBMonitor
                                                                                             Process: com.admobilize.matrix.app.android.usbcamera, PID: 19160
                                                                                             java.lang.UnsupportedOperationException: open failed:result=-99
                                                                                                 at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                                                                 at app.android.usbcamera.detection.USBCameraManager.handleOpen(USBCameraManager.java:77)
                                                                                                 at app.android.usbcamera.detection.USBCameraManager.handleConnect(USBCameraManager.java:106)
                                                                                                 at app.android.usbcamera.detection.USBCameraDevice$2.onConnect(USBCameraDevice.java:83)
                                                                                                 at com.serenegiant.usb.USBMonitor$3.run(USBMonitor.java:570)
                                                                                                 at android.os.Handler.handleCallback(Handler.java:733)
                                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                                 at android.os.Looper.loop(Looper.java:136)
                                                                                                 at android.os.HandlerThread.run(HandlerThread.java:61)
```

evgenek11, 2017-10-13:
@hpsaturn looks like your USB connection did not get proper Android permissions.

I am running usbCameraTest0 example, and I have similar error when Android drops the USB permissions (after sleeping, for example).

Anyway, your libusb library works OK and the error you've got is not related to the topic

hpsaturn, 2017-10-18:
But I changed all .so libraries to old version and all works again. The implementation and permissions flow seems be are ok. (Compiling with old NDK not worked for me)

Timmoth, 2017-11-07:
Brilliant @evgenek11 changing  
int len = usbi_parse_descriptor(buffer, "bb", &header, 0); 
worked perfectly for me however changing the optimization level however did not.

freedy-gb, 2018-02-06:
uncomment "NDK_TOOLCHAIN_VERSION := 4.9" from "libuvccamera/src/main/jni/Application.mk" , then it work

saurabh111091, 2018-02-26:
This worked for me too. You have been a savior!

TimberBug, 2018-03-02:
Hello, I changed the version of NDK to r14b, but there was a mistake. How should I solve this problem?

com.serenegiant.usbcameratest E/UVCCamera: [5801*UVCCamera.cpp:172:connect]:could not open camera:err=-1

TimberBug, 2018-03-02:
Then I changed the NDK version to r15c, which became the following error:  
    com.serenegiant.usbcameratest E/UVCCamera: [8871*UVCCamera.cpp:172:connect]:could not open camera:err=-12

UbuntuLover, 2018-03-09:
Hi guys. I rebuild this project with the latest ndk version 16. It works.

liabixiaoxin, 2018-04-19:
I experienced the same issue here and recompiling the .apk with NDK14 solved the problem.

paipeng, 2018-05-29:
Hello，

we have tried compiling with SDK 27 and NDK 17 and it works again by using "NDK_TOOLCHAIN_VERSION := 4.9"

davidhowe, 2018-10-22:
If you compile with NDK 17 you will need to change APP_ABI := armeabi-v7a x86 in Application.mk as described in https://github.com/saki4510t/UVCCamera/issues/358

tang5188, 2018-12-07:
https://github.com/saki4510t/UVCCamera/issues/181#issuecomment-335070530
https://github.com/saki4510t/UVCCamera/issues/181#issuecomment-335543365
hero！！！

funnywwh, 2019-04-17:
> @ThomasCZ Yes. 100% working with NDK 14b

ddin't work

SeanLuang, 2020-03-28:
It works！Thanks！