saki4510t/UVCCamera #182: Unable to open Camera of logitech C920 USB camera

himanshueinfochip, 2017-06-14:
I am facing below crash log in my app : 

```` Java
java.lang.UnsupportedOperationException: open failed:result=-99
	at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
	at com.camera.application.activity.WelcomeCameraActivity$5$1.run(WelcomeCameraActivity.java:414)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:148)
	at android.os.HandlerThread.run(HandlerThread.java:61)
````

What could be issue here any idea ?

DeepakDonde, 2017-06-30:
I am facing the same issue..did u find any solution?

freedy-gb, 2017-11-21:
same issue

wangkang6, 2018-05-09:
I have the same problem. Have you solved it?