saki4510t/UVCCamera #184: usbCameraTest Crashed while open uvc device on Google nexus 7 tablet

andymatos, 2017-06-19:
**I am trying to run example usbCameraTest  on my nexus 7 tablet which running  android 5.1.1.
 It shows USB_DEVICE_ATTACHED when I plug in my uvc camera. It's also works fine on getting device name :/dev/bus/usb/002/002.
But when I try to open the device for preview, the whole app crashed.
The logs are as follows:**

06-19 19:37:55.464 8073-8093/com.serenegiant.usbcameratest D/MainActivity: /dev/bus/usb/002/002
06-19 19:40:26.084 8073-8073/com.serenegiant.usbcameratest D/MainActivity: show camera dialog
06-19 19:40:27.387 8073-8093/com.serenegiant.usbcameratest D/UsbDeviceConnectionJNI: close
06-19 19:40:27.387 8073-8093/com.serenegiant.usbcameratest I/USBMonitor: name=/dev/bus/usb/002/002,desc=55,busnum=2,devnum=2,rawDesc=[B@29ea7f78
06-19 19:40:27.444 8073-8092/com.serenegiant.usbcameratest W/linker: libjpeg-turbo1500.so: unused DT entry: type 0x6ffffffe arg 0x2ef8
06-19 19:40:27.444 8073-8092/com.serenegiant.usbcameratest W/linker: libjpeg-turbo1500.so: unused DT entry: type 0x6fffffff arg 0x1
06-19 19:40:27.445 8073-8092/com.serenegiant.usbcameratest W/linker: libusb100.so: unused DT entry: type 0x6ffffffe arg 0x26e0
06-19 19:40:27.445 8073-8092/com.serenegiant.usbcameratest W/linker: libusb100.so: unused DT entry: type 0x6fffffff arg 0x1
06-19 19:40:27.449 8073-8092/com.serenegiant.usbcameratest W/linker: libuvc.so: unused DT entry: type 0x6ffffffe arg 0x33ec
06-19 19:40:27.449 8073-8092/com.serenegiant.usbcameratest W/linker: libuvc.so: unused DT entry: type 0x6fffffff arg 0x1
06-19 19:40:27.452 8073-8092/com.serenegiant.usbcameratest W/linker: libUVCCamera.so: unused DT entry: type 0x6ffffffe arg 0x5280
06-19 19:40:27.452 8073-8092/com.serenegiant.usbcameratest W/linker: libUVCCamera.so: unused DT entry: type 0x6fffffff arg 0x2
06-19 19:40:27.462 8073-8092/com.serenegiant.usbcameratest D/UsbDeviceConnectionJNI: close
06-19 19:40:27.463 8073-8092/com.serenegiant.usbcameratest I/libusb/core: [8092*core.c:2085:libusb_init2]:libusb v1.0.19.10903
06-19 19:40:27.463 8073-8092/com.serenegiant.usbcameratest I/libusb/usbfs: [8092*android_usbfs.c:504:op_init2]:start up hotplug event handler
06-19 19:40:27.463 8073-8092/com.serenegiant.usbcameratest I/libusb/usbfs: [8092*android_usbfs.c:512:op_init2]:call android_scan_devices
06-19 19:40:27.466 8073-8092/com.serenegiant.usbcameratest E/UVCCamera: [8092*UVCCamera.cpp:172:connect]:could not open camera:err=-1
                                                                        
                                                                        --------- beginning of crash
06-19 19:40:27.470 8073-8092/com.serenegiant.usbcameratest E/AndroidRuntime: FATAL EXCEPTION: BaseActivity
                                                                             Process: com.serenegiant.usbcameratest, PID: 8073
                                                                             java.lang.UnsupportedOperationException: open failed:result=-1
                                                                                 at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                                                 at com.serenegiant.usbcameratest.MainActivity$2$1.run(MainActivity.java:145)
                                                                                 at android.os.Handler.handleCallback(Handler.java:739)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                 at android.os.Looper.loop(Looper.java:135)
                                                                                 at android.os.HandlerThread.run(HandlerThread.java:61)
06-19 19:40:27.471 475-1512/? W/ActivityManager:   Force finishing activity 1 com.serenegiant.usbcameratest/.MainActivity
06-19 19:40:27.922 8073-8073/com.serenegiant.usbcameratest D/UsbDeviceConnectionJNI: close
06-19 19:40:28.528 689-689/? W/PackageManager: Failure retrieving resources for com.serenegiant.usbcameratest: Resource ID #0x0

It seems to be something wrong with opening the device, but my uvc camera works fine with other usbcamera app from google play store.
I am a newbee on using uvc device on android and don't know how to start  solving this problem.
Appreciate on any tips on debugging!!

RogerHardiman, 2017-06-19:
Can you check what resolutions the camera supports.
All the samples are written to only work if the camera supports 640x480 image resolution (along with its other resolutions). I have some cameras that are widescreen and do 640x400 (or higher widescreen) and had to modify the samples so the .Open() function passed in a different image size.

I have a fork of the project where one of the samples checks (USBTest0) checks the sizes before opening the UVC device.

andymatos, 2017-06-19:
Oh, it works fine with NDK 14! Thanks a lot!

RogerHardiman, 2017-06-19:
That is great news. It may be worth closing this PR as it is a duplicate of #181 

Thanks

closed 2017-06-20 by andymatos