saki4510t/UVCCamera #185: How to get yuv raw data？

andymatos, 2017-06-21:
Now I am working on an image process project using usb camera on android.
UVCCamera samples can show the preview of my usb camera module. The preview doesn't look right, since my usb camera module is a little bit special. My camera module consists of two cameras and using only one usb interface to transfer yuv stream to my pad.
For example, if each camera resolution is 640 * 480, it will be combined into 1280 * 960 in the hardware part. I need to divide it into two 640 * 480 manually.
So what I would like to know is that how can I access the raw data (in yuv fomat) from the usb interface and then I can convert these data into the normal form by myself.
I noticed the OpenCVwithUVC repo, but I still cannot find out the way to access yuv raw data?
Is there an api in libuvccamera or I should look deep into libuvc?

Any suggestions will be appreciated!!

closed 2017-07-02 by andymatos

FanHongchuang, 2017-12-06:
Are you get YUV raw data?

MTinside, 2025-08-12:
how were you able to resolve it?
thanks