saki4510t/UVCCamera #186: When we click a picture from the camera....then on returning to previous activity app freezes and after some secs crashes throwing following logs.

yuvraj24, 2017-06-22:
E/BufferQueueProducer: [unnamed-10193-1] dequeueBuffer: BufferQueue has been abandoned
E/Surface: dequeueBuffer failed (No such device)
E/BufferQueueProducer: [unnamed-10193-1] dequeueBuffer: BufferQueue has been abandoned
E/Surface: dequeueBuffer failed (No such device)
E/BufferQueueProducer: [unnamed-10193-1] dequeueBuffer: BufferQueue has been abandoned
E/Surface: dequeueBuffer failed (No such device)
E/BufferQueueProducer: [unnamed-10193-1] dequeueBuffer: BufferQueue has been abandoned
E/Surface: dequeueBuffer failed (No such device)
E/BufferQueueProducer: [unnamed-10193-1] dequeueBuffer: BufferQueue has been abandoned
E/Surface: dequeueBuffer failed (No such device)
Explicit concurrent mark sweep GC freed 79977(5MB) AllocSpace objects, 31(8MB) LOS objects, 28% free, 40MB/56MB, paused 1.041ms total 158.049ms

dinhtung249, 2022-09-24:
Did your find the resolve? Can you share it?
tks

zipswich, 2023-11-16:
Same problem here.
       E  [SurfaceTexture-1-11549-61] dequeueBuffer: BufferQueue has been abandoned
       E  dequeueBuffer failed (No such device)