saki4510t/UVCCamera #191: Get RBG image from NV21 frame

Osarez7, 2017-07-05:
Hi , I'm working in an app and I need to get two types of images from the camera frame, one is a byte array in NV21 format and the other one is a  RGB bitmap

I'm using the IFrameCallback to get the frame from the camera
`mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_NV21);`

Then I get the byte array that I need 
```java
byte[] mFrame;
private final IFrameCallback mIFrameCallback = new IFrameCallback() {
    @Override
    public void onFrame(final ByteBuffer frameBuffer) {
           mFrame = new byte[frameBuffer.remaining()];
           frameBuffer.get(mFrame, 0, mFrame.length);
     }
}
```
In other method I get the bitmap 

```java
public Bitmap getBitmap(){
    if (mFrame != null) {
                YuvImage yuv = new YuvImage(mFrame, ImageFormat.NV21, mFrameWidth, mFrameHeight, null);
                ByteArrayOutputStream os = new ByteArrayOutputStream();
                yuv.compressToJpeg(new Rect(0, 0, mFrameWidth, mFrameHeight), 100, os);
                byte[] bytes = os.toByteArray();
                return  BitmapFactory.decodeByteArray(bytes, 0, bytes.length);
            }
}
```

The bitmap that I'm getting has a wrong format, for example in this image the object has to be orange instead of blue

[image]

Any advice?

FanHongchuang, 2017-12-07:
@Osarez7 
Hello,
Are you resolve it?

gaborbencebekesi, 2018-04-10:
Hello @Osarez7,

Did you solve this problem?

Thanks,
Gabor

capsane, 2018-12-29:
hi, when I use PIXEL_FORMAT_NV21, the format of the picture was wrong, which swapped the red and blue channel.
In the end, I change the format into PIXEL_FORMAT_YUV420SP like this:
`mUVCCamera.setFrameCallback(callback, UVCCamera.PIXEL_FORMAT_YUV420SP);`
then I get the rigth picture.
btw, try using RenderScript and ScriptIntrinsicYuvToRGB to accelerate the transforming.

janoonaj, 2019-01-24:
Yes, there is a problem with the enums on the library. If you want N21 you need to write YUV420SP

More here: [](https://github.com/saki4510t/UVCCamera/issues/207)

liqvip, 2022-07-05:
@Osarez7 Your code showed me how to convert ByteBuffer to a byte array, thank you very much!