saki4510t/UVCCamera #192: Blank? Glitch images occur sometimes

sappho192, 2017-07-06:
Sometimes blank(or glitch) images with partial grey coloured area and partial images through camera occur. Is this out of performance issue? Does anyone experienced this?
My phone is Samsung Galaxy S7 and the camera is dual-lens camera from here: https://www.aliexpress.com/item/Synchronization-USB-3-0-MJPEG-30fps-2560-720-OV9714-CMOS-mini-webcam-Dual-lens-usb-camera/32642278164.html?spm=2114.13010608.0.0.z4YcXw
Thanks.

RogerHardiman, 2017-07-21:
I get the same problem too with a single camera uvc camera and have seen it on several different handselts. I think it must be down to the volume of data on the USB bus.
I did try switching from YUYV format (raw data) to the MJPEG modes (which than incurs CPU time to decompress the JPEGs) but from memory both had occasional glitches.

obg2sens, 2017-07-25:
Maybe it is same issue as this https://github.com/saki4510t/UVCCamera/issues/127

sappho192, 2017-08-02:
@RogerHardiman @obg2sens What's the resolution of your camera video settings?

RogerHardiman, 2017-08-03:
640x400  (this is why I modified one of the example programs to select a valid resolution instead of being hard coded to 640x480)

closed 2020-02-17 by sappho192