saki4510t/UVCCamera #195: Sometimes preview freeze on some devices

ldima, 2017-07-12:
Preview work good, but sometimes it freeze.
Log is clear.  It has on only HTC One and Sony C6903. 
On nexus 5x, samsung s7, samsung note 4 it 

saki4510t did you see similar bugs? What advice could you please give me?
Tnx.

sappho192, 2017-07-13:
Me too on Galaxy S7, with USBCameraTest0.

ldima, 2017-07-13:
If I increase preview resolution I faster get frozen preview on all device.
I use nexus 5 and execute usbCameraTest0 from examples.
Preview freeze with 100%, Time when preview will froze depends on resolution.
If I use 640*480 preview work Great.

ldima, 2017-07-13:
I switch On Logs in USBMonitor, UVCCamera, but it didn't result.

07-13 16:14:38.790 10993-11030/com.serenegiant.usbcameratest0 I/USBMonitor: UsbControlBlock:constructor
07-13 16:14:39.124 10993-11030/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-13 16:14:39.124 10993-11030/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/002,desc=40,busnum=1,devnum=2,rawDesc=[B@6f49d1b
07-13 16:14:39.124 10993-11030/com.serenegiant.usbcameratest0 V/MainActivity: onConnect:
07-13 16:14:39.146 10993-11022/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-13 16:14:39.146 10993-11022/com.serenegiant.usbcameratest0 I/libusb/core: [11022*core.c:2085:libusb_init2]:libusb v1.0.19.10903
07-13 16:14:39.146 10993-11022/com.serenegiant.usbcameratest0 I/libusb/usbfs: [11022*android_usbfs.c:504:op_init2]:start up hotplug event handler
07-13 16:14:39.146 10993-11022/com.serenegiant.usbcameratest0 I/libusb/usbfs: [11022*android_usbfs.c:512:op_init2]:call android_scan_devices
07-13 16:14:39.170 10993-11022/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480"]},{"index":2,"type":6,"default":1,"size":["1280x720","1920x1080","2304x1296","2304x1536"]}]}
07-13 16:14:39.174 10993-11237/com.serenegiant.usbcameratest0 I/libUVCCamera: [11237*UVCPreview.cpp:492:prepare_preview]:frameSize=(2304,1536)@MJPEG
07-13 16:14:39.175 10993-11239/com.serenegiant.usbcameratest0 I/libUVCCamera: [11239*UVCPreview.cpp:258:callbackPixelFormatChanged]:PIXEL_FORMAT_RAW:
07-13 16:14:44.476 10993-11238/com.serenegiant.usbcameratest0 W/libUVCCamera: [11238*UVCPreview.cpp:119:get_frame]:allocate new frame
07-13 16:14:44.477 10993-11237/com.serenegiant.usbcameratest0 W/libUVCCamera: [11237*UVCPreview.cpp:119:get_frame]:allocate new frame
07-13 16:14:44.507 10993-11238/com.serenegiant.usbcameratest0 W/libUVCCamera: [11238*UVCPreview.cpp:119:get_frame]:allocate new frame
07-13 16:14:44.541 10993-11238/com.serenegiant.usbcameratest0 W/libUVCCamera: [11238*UVCPreview.cpp:119:get_frame]:allocate new frame
07-13 16:14:44.574 10993-11238/com.serenegiant.usbcameratest0 W/libUVCCamera: [11238*UVCPreview.cpp:119:get_frame]:allocate new frame

After that preview was frozen.

jerry0864, 2017-07-13:
@ldima could you tell me how to set resolution? thanks..

sappho192, 2017-07-13:
That kind of symptom highly seems to be memory issue.
And your log says "**allocate** new frame" but there's no log like "free new frame", right?
Maybe there're freeing code but just the log informing that is not exist.
Debug your app and check if the RAM memory which the app consume  is continuously increasing.

ldima, 2017-07-13:
It were all logs that I have
What buffer I can increase for that?

ldima, 2017-07-13:
There are 2 threads in UVCPreview.cpp
1 thread add frame to array 
2 thread wait and get frame from array

Preview started, and 1 thread wil add frame to array, 2 thread will wait a new frame but frame wont add and preview will freeze. It happen after a few second work of preview.

ldima, 2017-07-24:
Current library don't work still on some device with HD and FullHd (for example: nexus 5, HTC One, Sony C6903.). I think error in C or CPP code.
Frames stop in one moment. If you have power phone as samsung s7 or modern xiaomi you won't see this problem.

ldima, 2017-07-27:
When was bug ,preview got offset . It seen on picture 
[image]

ldima, 2017-07-31:
Could particular camera throw broken or specified frame and  libuvc stop.
I tried to add logs in stream.c and uvcpreview.cpp by I have not found answer (what is problem) yet

gaborbencebekesi, 2018-05-03:
Is there any solution for this problem?
Thanks!

805447391, 2018-05-14:
@gaborbencebekesi Sure.
Here is my stub solution:
First, using USBMonitor and UVCCamera implement a module witch can get frame data from uvc stream, and then, using OpenCV or JavaCv or Pure Java or so many what you liked to convert pixels array to Android Bitmap, finally, we just need to using ImageView.setImage() to render it.
By the way, open Hard Acc will got high performance in this way I just metioned.
But this way is always not the best way to render frame than OpenGL and others.