saki4510t/UVCCamera #197: Change Preview size of cameraview

aviontechnology, 2017-07-13:
Hi,

We got a big tablet (32 inches), need a cameraview to fit into the Tab. We are using this code for camera preview
mUSBMonitor = new USBMonitor(this, mOnDeviceConnectListener);
mCameraHandler = UVCCameraHandler.createHandler(this, mUVCCameraView,
USE_SURFACE_ENCODER ? 0 : 1, PREVIEW_WIDTH, PREVIEW_HEIGHT, PREVIEW_MODE);

To set the preview size using following code
private UVCCamera camera;
camera.setPreviewSize(960, 1280);//width , height

But the App is crashing and generating 
"Caused by: java.lang.IllegalArgumentException: Failed to set preview size"

Can you please suggest if something else needs to be changed to increase camera preview size.

Thanks!

sappho192, 2017-07-13:
You need to get a list of available preview sizes which **your camera supports**.
You can check according code in somewhere in sample codes.

aviontechnology, 2017-07-14:
Can you please let us know the method to get list of available preview sizes.
Is there any sample code.

sappho192, 2017-07-14:
Use getSupportedSize() in UVCCamera Class.
Check this code: [https://github.com/saki4510t/UVCCamera/pull/175/files/3c18d3dce6caefe5a87bf1637396ef93fe8b4f8f..1b98d47d3d7bb0c99816b37fa79b930db7d7587d](url)

aviontechnology, 2017-07-14:
Thank you for your kind help .

I am using camera handler as per usb camera Test8 
mCameraHandler.open(ctrlBlock);
there is any thing to set preview with camera handler.

aviontechnology, 2017-07-14:
Also, I need to grab the highest possible resolution of the image. Can you please let me know where to mention the resolution.

Thanks!

sappho192, 2017-07-18:
If you just add those codes in [https://github.com/saki4510t/UVCCamera/pull/175/files](https://github.com/saki4510t/UVCCamera/pull/175/files) then you will get highest resolution from the camera. Did you try that on CameraTest8?

closed 2017-08-24 by aviontechnology