saki4510t/UVCCamera #198: FATAL EXCEPTION

dalwadi2, 2017-07-14:
The app is working completely fine before, I Don't know now its stop working and giving me this type of error.
```

FATAL EXCEPTION: UVCCameraCapturer
                                                              Process: com.twilio.video.examples.uvccameracapturer, PID: 23779
                                                              java.lang.UnsupportedOperationException: open failed:result=-1
                                                                  at com.serenegiant.usb.UVCCamera.open(UVCCamera.java:204)
                                                                  at com.twilio.video.examples.uvccameracapturer.UVCCameraCapturer.startUvcCameraPreviewOnCameraThread(UVCCameraCapturer.java:198)
                                                                  at com.twilio.video.examples.uvccameracapturer.UVCCameraCapturer.access$500(UVCCameraCapturer.java:30)
                                                                  at com.twilio.video.examples.uvccameracapturer.UVCCameraCapturer$2$2.run(UVCCameraCapturer.java:85)
                                                                  at android.os.Handler.handleCallback(Handler.java:751)
                                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                  at android.os.Looper.loop(Looper.java:154)
                                                                  at android.os.HandlerThread.run(HandlerThread.java:61)
```
Can you help me to find out the solution.?

sappho192, 2017-07-14:
Which NDK version are you using? If it's 15, try 14b.

dalwadi2, 2017-07-14:
okay, i'll try your suggestion.
if it will succeed, I'll close the issue.
thanks @sappho192

dalwadi2, 2017-07-18:
yes, it's working on 14b @sappho192 .
can you Explain what was the issue with updated NDK?

closed 2017-07-18 by dalwadi2

sappho192, 2017-07-18:
This issue is in same context of Issue #181 .
Don't know what the problem exactly is, but it seems that Linux system call related stuff in NDK 15 have some matter. Maybe next release of NDK will solve this but not sure...

dalwadi2, 2017-07-18:
okay, thanks @sappho192

sappho192, 2017-07-22:
Please close this issue since it's solved :)

dalwadi2, 2017-07-25:
already I closed it @sappho192