saki4510t/UVCCamera #201: Image resolution

aviontechnology, 2017-07-15:
Hi,

I am following these instructions to change image resolution and did required changes in the code. It's not applying the changes (changing image to 1920*1080). 
Currently the image gets stored in 1020*764. 
https://github.com/saki4510t/UVCCamera/wiki/howto_change_resolution

Using "usbCameraTest8" sample, also just to let you know I am running this code on 24" and 32" tablets.  

Thanks!

sappho192, 2017-07-18:
You've opened duplicated issue like #197  and #199 .
It's enough to discuss this in #197 . It would be nice of you if you close #199 and this(#201 ).

aviontechnology, 2017-08-24:
i think image resolution  and camera preview quality issue is still open.
Can you please suggest what all i need to do to improve image quality and camera preview quality.

sappho192, 2017-08-24:
There are options that your camera provides in hardware level.
Example, my camera provides 1280*480@20fps or 30fps, 2560*960@20fps or 30fps / MJPEG, YUV format.
Image quality can be better if you choose higher resolution for your camera in your code, but it can drop performance.
Camera preview quality(fps or else) would be changed if you modify libuvccamera library code. Especially  handling and passing frame data from camera.
I don't know exact way to do that since I'm not the creator of this project and just know little bit more than others by searching gathering info on here.

aviontechnology, 2017-08-24:
How can i do this "if you choose higher resolution for your camera in your code"

sappho192, 2017-08-24:
Check code change in #175 .

aviontechnology, 2017-08-24:
i already use it but now showing getting good quality resolution.