saki4510t/UVCCamera #202: codecInfo.getCapabilitiesForType(mimeType) creates IllegalArgumentException sometimes

jp1017, 2017-07-17:
Any advice?

sappho192, 2017-07-18:
It would be nice of you if you provide detailed information like log, situation, related code.

jp1017, 2017-07-19:
@sappho192 Thanks for reply.

error stack here: 

```
07-19 08:49:43.464 32219-32285 W/System.err: java.lang.IllegalArgumentException
07-19 08:49:43.464 32219-32285 W/System.err:     at android.media.MediaCodecList.getCodecCapabilities(Native Method)
07-19 08:49:43.465 32219-32285 W/System.err:     at android.media.MediaCodecInfo.getCapabilitiesForType(MediaCodecInfo.java:291)
07-19 08:49:43.465 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.encoder.MediaSurfaceEncoder.selectColorFormat(MediaSurfaceEncoder.java:209)
07-19 08:49:43.466 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.encoder.MediaSurfaceEncoder.selectVideoCodec(MediaSurfaceEncoder.java:168)
07-19 08:49:43.466 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.encoder.MediaSurfaceEncoder.prepare(MediaSurfaceEncoder.java:89)
07-19 08:49:43.466 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.encoder.MediaMuxerWrapper.prepare(MediaMuxerWrapper.java:82)
07-19 08:49:43.467 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.service.CameraServer$CameraThread.handleStartRecording(CameraServer.java:579)
07-19 08:49:43.467 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.service.CameraServer$CameraThread.access$1100(CameraServer.java:387)
07-19 08:49:43.467 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.service.CameraServer.handleMessage(CameraServer.java:345)
07-19 08:49:43.468 32219-32285 W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:102)
07-19 08:49:43.468 32219-32285 W/System.err:     at android.os.Looper.loop(Looper.java:136)
07-19 08:49:43.468 32219-32285 W/System.err:     at com.inst.instcarlauncher.uvccamera.service.CameraServer$CameraThread.run(CameraServer.java:750)
```

sappho192, 2017-07-19:
It seems like some arguments passed from getCapabilitiesForType@MediaCodecInfo.java:291) to getCodecCapabilities is not valid. You need to set breakpoint on MediaCodecInfo.java:291 then find out and compare arguments with succeeded cases and failed cases.