saki4510t/UVCCamera #203: how to add the function of manually selecting the resolution?

kingjie0210, 2017-07-17:
Hi @saki4510t ,

I am relatively new to Android programming (about 1 months). I am working on an application in which I have a USB camera always connected to Android device. I need the following functions: recording videos, supporting manual selection of resolution. I found that your usbCameraTest8 can record video, modify the brightness, contrast. Do you have plans to add the function of manually selecting the resolution or do you have any suggestions on how to add the function of manually selecting the resolution?

with best regards,
Super King

jp1017, 2017-07-17:
Yes, you can try Seekbar

sappho192, 2017-07-18:
Refer this code changes [https://github.com/saki4510t/UVCCamera/pull/175/files](https://github.com/saki4510t/UVCCamera/pull/175/files) and try it.