saki4510t/UVCCamera #205: I CAN NOT OPEN IT WITH ANDROID STUDIO, Invalid VCS root mapping, NOT EASY MR. SAKI !!!

aulker62, 2017-07-20:
Your app is very difficult to open with Android Studio, there is always ERROR. 
Invalid VCS root mapping...

Mr Saki, why don't configure your gradile files for easy opening with Android Studio? I am downloading other peoples App's,  they are opening easily. But yours, super difficult.

You are UVC master Mr. Saki, but please look to the big picture.
,
Many thanks.

sappho192, 2017-07-20:
I'm sorry but I have no matter testing this project with Android Studio.
Just ignore VCS and install NDK 14b and then you are good to go.

RogerHardiman, 2017-07-21:
I use the library in Android Studio with no problems.
I probably used Git Clone to check out the source tree from the command line and then opened the project in Android Studio.

abhishekkumar1902, 2017-08-24:
I am trying to build a new project for API 23 where I copied exactly the code and layout that is present in https://github.com/saki4510t/UVCCamera/tree/master/usbCameraTest0 and then added a new module in the project for importing the generated "libuvccamera-release.aar" library. All the code seems to be compiling fine except at the lines where queueEvent() is used The compiler throws the following error:

Error:(403, 13) error: cannot find symbol method queueEvent(,int)

Any idea what I am missing in the new project?

jp1017, 2017-08-25:
gradle add this: 

```
repositories {
	maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
        jcenter()
    }
```

abhishekkumar1902, 2017-08-25:
I tried adding:

> repositories {
> 	maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
>         jcenter()
>     }

in gradle, but still not working, any help will be appreciated!

abhishekkumar1902, 2017-08-25:
I tried adding:

repositories {
maven { url 'http://raw.github.com/saki4510t/libcommon/master/repository/' }
jcenter()
}

in gradle, but still not working, any help will be appreciated!

abhishekkumar1902, 2017-08-25:
I observe that even though I have added libcommon v1.5.20 and libuvccamera.aar files as dependency to the project, the compiler is still unable to refer to the **queueEvent()** method which resides in **com.serenegiant.common.BaseActivity** class. Any idea why?

aulker62, 2017-08-26:
I think so, everybody use it personally. Not easy to integration with other applications. I tried to change all files package names one by one manually, it takes few hours also, but no positive result. No integration...

Mahdis3, 2018-02-06:
please check this link:
https://intellij-support.jetbrains.com/hc/en-us/community/posts/115000424850-Invalid-VCS-root-mappings-when-creating-project-from-version-contro
maybe it help u