saki4510t/UVCCamera #207: documentation error for NV21 and YUV420 modes

RogerHardiman, 2017-08-04:
Hi
This is a great library and works really well.

Small issue with the code and YUV420 and NV21 outputs from cameras in YUYV mode

In frame.c it are some comments that says that 'iyuv420sp' (first character is 'i') is for NV21.
'''
@brief Convert a frame to iyuv420sp(NV21)
'''

But in UVCPreview.cpp the PIXEL_FORMAT_NV21 does not call 'iyuv420sp'. It calls a different function
'''
	  case PIXEL_FORMAT_YUV20SP:
		LOGI("PIXEL_FORMAT_YUV20SP:");
		mFrameCallbackFunc = uvc_yuyv2iyuv420SP;
		callbackPixelBytes = (sz * 3) / 2;
		break;
	  case PIXEL_FORMAT_NV21:
		LOGI("PIXEL_FORMAT_NV21:");
		mFrameCallbackFunc = uvc_yuyv2yuv420SP; // <- this does not match frame.c comments
		callbackPixelBytes = (sz * 3) / 2;
		break;
'''

I'll try and look at the output data and then see if the code is wrong or if it is just the comments that are wrong.

Thanks again for the library. It is really good.

fengbenpaao, 2017-08-08:
do you know how to show h264 preview ?  Let's  learn from each other!!!

RogerHardiman, 2017-08-08:
I do not know how to get the video in H264 format from the camera.

sappho192, 2017-08-24:
Was the output data different? 
I think this is his mistake since when I load frame data with NV21, red and blue data order is swapped so I need to swap them. AFAIK reversed RGB order of NV21 is YUV20SP.

RogerHardiman, 2017-08-27:
Hi @sappho192
If I remember correctly the yuv data returned does not match Enum name.

sappho192, 2017-08-27:
Then I will try to modify the code if it's his mistake.

RogerHardiman, 2017-08-27:
We need to watch that existing users will have written their code around the incorrect functionality. So we may want some extra enums.

wlxyhy, 2021-06-22:
I found the same problem, nv21 and 420sp should be changed.  
in UVCPreview.cpp
` case PIXEL_FORMAT_YUV20SP:
		LOGI("PIXEL_FORMAT_YUV20SP:");
		mFrameCallbackFunc = uvc_yuyv2iyuv420SP;
		callbackPixelBytes = (sz * 3) / 2;
		break;
	  case PIXEL_FORMAT_NV21:
		LOGI("PIXEL_FORMAT_NV21:");
		mFrameCallbackFunc = uvc_yuyv2yuv420SP;
		callbackPixelBytes = (sz * 3) / 2;
		break;`

while in frame.c
`uvc_error_t uvc_yuyv2iyuv420SP(uvc_frame_t *in, uvc_frame_t *out) {
	......
	for (h = 0; h < hh - 1; h += 2) {
		uint8_t *y0 = dest + width * h;
		uint8_t *y1 = y0 + width;
		const uint8_t *yuv = src + src_width * h;
		for (w = 0; w < width; w += 4) {
			*(y0++) = yuv[0];	// y
			*(y0++) = yuv[2];	// y'
			*(y0++) = yuv[4];	// y''
			*(y0++) = yuv[6];	// y'''
			*(uv++) = yuv[3];	// v
			*(uv++) = yuv[1];	// u
			*(uv++) = yuv[7];	// v
			*(uv++) = yuv[5];	// u
			*(y1++) = yuv[src_width+0];	// y on next low
			*(y1++) = yuv[src_width+2];	// y' on next low
			*(y1++) = yuv[src_width+4];	// y''  on next low
			*(y1++) = yuv[src_width+6];	// y'''  on next low
			yuv += 8;	// (1pixel=2bytes)x4pixels=8bytes
		}
	}`
'yyyyvuvu', it's obviously NV21