saki4510t/UVCCamera #208: Is there anyone who converted frame data to ARGB8888 to manipulate them?

sappho192, 2017-08-08:
Hi,

I'm going to process each frame data in native C side, so it is necessary to convert the format of each data.
Currently I can achieve RGB565 or NV21 formatted frames so tried to convert them to ARGB8888 but it failed.
The reference of RGB565 -> ARGB8888 is nearly don't exist and NV21 -> ARGB8888 is [here](https://github.com/arrayfire/androidcl/blob/master/assets/kernels.cl) but it doesn't work on this project.
Am I missing something?

Any advice can help me a lot.

sappho192, 2017-08-15:
Solved it.

Check this repository. https://github.com/arrayfire/androidcl/blob/master/assets/kernels.cl

closed 2017-08-15 by sappho192

vongohren, 2025-02-03:
This is a while ago, but you did not find any other ways?