saki4510t/UVCCamera #209: Stuck at nativeStopPreview

afvilalva, 2017-08-08:
I have an issue where I need to disable câmera preview and them it get stuck and since It's being called from UI thread an ANR pops up on screen.
Looking into traces.txt it seems that the point where it hapens is in UVCPreview::stopPreview in a pthread join.
Traces.txt was attached to this issue

DeviceModel: Nexbox-A95X
Android Version: 6.0.1 (API 23)

"main" prio=5 tid=1 Native
  | group="main" sCount=1 dsCount=0 obj=0x74d6efb0 self=0xf5536500
  | sysTid=11136 nice=0 cgrp=default sched=0/0 handle=0xf772ab34
  | state=S schedstat=( 9549469868 8834260543 56989 ) utm=606 stm=348 core=0 HZ=100
  | stack=0xff100000-0xff102000 stackSize=8MB
  | held mutexes=
  kernel: __switch_to+0x74/0x8c
  kernel: futex_wait_queue_me+0xd8/0x164
  kernel: futex_wait+0xec/0x23c
  kernel: do_futex+0x1a4/0xa04
  kernel: compat_SyS_futex+0x94/0x170
  kernel: cpu_switch_to+0x48/0x4c
  native: #00 pc 00017b9c  /system/lib/libc.so (syscall+28)
  native: #01 pc 00040295  /system/lib/libc.so (pthread_join+124)
  native: #02 pc 0000eed0  /data/app/com.contartec.contagente-1/lib/arm/libUVCCamera.so (_ZN10UVCPreview11stopPreviewEv+56)
  native: #03 pc 000072b4  /data/app/com.contartec.contagente-1/lib/arm/libUVCCamera.so (_ZN9UVCCamera11stopPreviewEv+16)
  native: #04 pc 0022c47d  /data/app/com.contartec.contagente-1/oat/arm/base.odex (Java_com_serenegiant_usb_UVCCamera_nativeStopPreview__J+88)
  at com.serenegiant.usb.UVCCamera.nativeStopPreview(Native method)
  at com.serenegiant.usb.UVCCamera.stopPreview(UVCCamera.java:429)
  - locked <0x00d18b19> (a com.serenegiant.usb.UVCCamera)
  at com.serenegiant.usb.UVCCamera.close(UVCCamera.java:237)
  - locked <0x00d18b19> (a com.serenegiant.usb.UVCCamera)
  at com.serenegiant.usb.UVCCamera.destroy(UVCCamera.java:437)
  - locked <0x00d18b19> (a com.serenegiant.usb.UVCCamera)
  at com.kajoo.views.UvcCameraView.releaseCamera(UvcCameraView.java:208)
  - locked <0x0ff007de> (a com.kajoo.views.UvcCameraView)
  at com.kajoo.views.UvcCameraView$MyDeviceConnectListener.onDisconnect(UvcCameraView.java:380)
  at com.serenegiant.usb.USBMonitor$UsbControlBlock.close(USBMonitor.java:1313)
  - locked <0x0f4d2dbf> (a com.serenegiant.usb.USBMonitor$UsbControlBlock)
  at com.serenegiant.usb.USBMonitor.destroy(USBMonitor.java:144)
  at com.kajoo.views.UvcCameraView.unregisterUsbMonitor(UvcCameraView.java:84)
  at com.kajoo.views.UvcCameraView.disconnectCamera(UvcCameraView.java:247)
  at com.kajoo.views.KajooCameraBridgeViewBase.onExitStartedState(KajooCameraBridgeViewBase.java:371)
  at com.kajoo.views.KajooCameraBridgeViewBase.processExitState(KajooCameraBridgeViewBase.java:341)
  at com.kajoo.views.KajooCameraBridgeViewBase.checkCurrentState(KajooCameraBridgeViewBase.java:313)
  at com.kajoo.views.KajooCameraBridgeViewBase.disableView(KajooCameraBridgeViewBase.java:242)
  - locked <0x0886dc8c> (a java.lang.Object)
  at com.contartec.contagente.MainActivity$InterfaceHandler.handleMessage(MainActivity.java:489)
  at android.os.Handler.dispatchMessage(Handler.java:102)
  at android.os.Looper.loop(Looper.java:148)
  at android.app.ActivityThread.main(ActivityThread.java:5422)
  at java.lang.reflect.Method.invoke!(Native method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

[traces.txt](https://github.com/saki4510t/UVCCamera/files/1208821/traces.txt)

Mapperyang, 2019-07-03:
Hello, did you have resolved it?

Leessy, 2019-07-04:
stopPreview() there ANR.  Is that ok?   Android Version: 7.1

Jimmy-J, 2022-07-14:
Hello, did you have resolved it?

Jimmy-J, 2022-07-20:
currently stable https://github.com/jiangdongguo/AndroidUSBCamera/issues/422

afvilalva, 2022-07-20:
> currently stable jiangdongguo/AndroidUSBCamera#422

Hello Jimmy,
I didn't resolve this issue that time we just changed the board we were
using (as far I remember).
Unfortunately I don't have access to this setup anymore.

Thx for your reply