saki4510t/UVCCamera #210: The UVCCamera project can't work correctly after update ndk to 15.1 in Android Studio

FanHongchuang, 2017-08-15:
The UVCCamera project can't work correctly after update ndk to 15.1 in Android Studio.

The message is:
`04-09 08:10:44.441 1747-1767/com.serenegiant.usbcameratest8 E/UVCCamera: [1767*UVCCamera.cpp:172:connect]:could not open camera:err=-1`

sappho192, 2017-08-15:
https://github.com/saki4510t/UVCCamera/issues/181
You need to use this on NDK14b.

abhishekkumar1902, 2017-08-24:
I am trying to build a new project for API 23 where I copied exactly the code and layout that is present in https://github.com/saki4510t/UVCCamera/tree/master/usbCameraTest0 and then added a new module in the project for importing the generated "libuvccamera-release.aar" library. All the code seems to be compiling fine except at the lines where queueEvent() is used The compiler throws the following error:

Error:(403, 13) error: cannot find symbol method queueEvent(,int)

Any idea what I am missing in the new project?