saki4510t/UVCCamera #213: Rotation on cameraview

mumtazcem, 2017-08-16:
Any idea about how can I rotate the view 180 degrees in usbcameratest7? I dont know exactly what should I do. I could rotate the matrix that I get from using getTransformMatrix(mtx) but I cant really load the matrix back to the SurfaceTexture.

Thanks in advance.

sappho192, 2017-08-16:
How about this code?
in **AndroidManifest.xml**
<activity
            android:name=".LiveFeatureActivity"
            android:label="@string/title_activity_live_feature"
            android:screenOrientation="landscape"> ...

mumtazcem, 2017-08-16:
I have a button that will rotate the view whenever I want, so I cant use this code in manifest

sappho192, 2017-08-24:
Then I think you need to transform each frame data.
In general, the screen is rotated when the device rotates but if you want to rotate it manually then you need to do it in code level.

aviontechnology, 2017-08-29:
I can possible to rotate camera preview with usbcameratest8 sample
I use below code 
		<com.serenegiant.widget.UVCCameraTextureView
			android:id="@+id/camera_view"
			android:layout_width="match_parent"
			android:layout_height="match_parent"
			android:layout_centerHorizontal="true"
			android:layout_centerVertical="true"
			android:background="@color/GRAY"
			android:scaleType="fitXY"
			android:layout_gravity="center"
			android:rotation="270"
			android:scaleY="-1"
			/>

Now i am trying to rotate camera preview in sample usbcameratest0. Please provide some guide line what all i need to use.

miggymamba, 2017-10-27:
@aviontechnology have you done this? anyone knows how to do this?

aviontechnology, 2017-10-27:
@miggymamba 
you can try with android:rotation=" ? "property 
eg.android:rotation="270"
try 90 ,180,270 or -90,-180,-270

and also try scaleY and scaleX combination 
it rotate in every angel now