saki4510t/UVCCamera #214: Camera Preview is not working with some webcam.

aviontechnology, 2017-08-16:
Hello,
I use UVCCamera to connect external webcam in my project.
But its not working with some webcam.
1.Logitech V-U0024 and 2.Microsoft Life Cam VX-700.

So How can i make it work with any webcam. 
Please give me some solution for this.

Thank You

sappho192, 2017-08-16:
1. Be sure you are not using NDK 15, but NDK 14b
2. https://github.com/saki4510t/UVCCamera/pull/175 apply this code change to get rid of resolution issue.
3. Provide your log or clues

aviontechnology, 2017-08-18:
Yes, i am using NDK 15
Pkg.Desc = Android NDK
Pkg.Revision = 15.1.4119039

Which NDK Version I need to use.

sappho192, 2017-08-18:
Use NDK 14b. You can manually assign the location of NDK in "Project Structure" in Android Studio.

aviontechnology, 2017-08-18:
Hello,

As you suggest i did changes with NDK and now i am using NDK 14b. 
I use usbCameraTest8's MainActivity.java in my project.
There is no 	**private UVCCamera mUVCCamera;** object.
In  usbCameraTest8not use UVCCamera
thats why i can not possible to use 2. [#175](https://github.com/saki4510t/UVCCamera/pull/175) apply.

So please give me one solution which work with **usbCameraTest8's MainActivity.java**

Thank You
Kiran Kirve

sappho192, 2017-08-18:
#175 's patch is getting best resolution that your camera supports.
If you know supported resolution of your camera, then just overwrite it to width and height variables.

aviontechnology, 2017-08-18:
I am looking solution which work for all web cams.

sappho192, 2017-08-19:
Then make the code like #175 .

closed 2017-08-24 by aviontechnology