saki4510t/UVCCamera #217: Add support to stream over rtmp

ManojMMWWIndia, 2017-08-23:
Need rtmp support for lib. Other RTMP lib using android native camera to get data from and UVCCamera is not working with them. Can you help on this?

sappho192, 2017-08-24:
The creator of this project have stopped replying on issues so I'm afraid that you need to develop it by yourself.

ManojMMWWIndia, 2017-10-25:
please tell me how do I get audio+video data in byte form so that I can stream that data

jp1017, 2017-10-26:
@ManojMMWWIndia 

You can get byte[] from this:

```
mUVCCamera.setFrameCallback(mIFrameCallback, UVCCamera.PIXEL_FORMAT_YUV);

		private final IFrameCallback mIFrameCallback = new IFrameCallback() {
			@Override
			public void onFrame(final ByteBuffer frame) {
				try {
						byte[] buf = new byte[frame.remaining()];
						frame.get(buf);

                                               //then we can get byte[]
					}
				} catch (Exception e) {
					KLog.e(TAG, e.getMessage());
				}

			}
		};
```

FanHongchuang, 2017-12-07:
@ManojMMWWIndia 
Hello,
Are you resolve it about RTMP?

OleksandrAndro, 2019-10-24:
Please, help to encode "ByteBuffer frame" to h264 byte[]