saki4510t/UVCCamera #219: Issue using uvccamera .aar in a new android project

abhishekkumar1902, 2017-08-24:
I am trying to build a new project for API 23 where I copied exactly the code and layout that is present in https://github.com/saki4510t/UVCCamera/tree/master/usbCameraTest0 and then added a new module in the project for importing the generated **"libuvccamera-release.aar"** library. All the code seems to be compiling fine except at the lines where **queueEvent()** is used The compiler throws the following error:

**Error:(403, 13) error: cannot find symbol method queueEvent(<anonymous Runnable>,int)**

Any idea what I am missing in the new project?

aulker62, 2017-08-25:
Same problem for me. Anybody can help please?
I want to combine my own project and UVCCamera together, but i can't find correct solution, i am trying more than one week.
Thanks for your help.

abhishekkumar1902, 2017-08-25:
I observe that even though I have added libcommon v1.5.20 and libuvccamera.aar files as depepdency to the project, the compiler is still unable to refer to the **queueEvent()** method which resides in **com.serenegiant.common.BaseActivity** class. Any idea why?