saki4510t/UVCCamera #224: Some Error happened when recording

xiaochongdeweiba, 2017-08-31:
Exception Log Time:[Thu Aug 31 09:46:46 CST 2017] [619.343101]

Exception Class: Native (NE)
Exception Type: SIGSEGV

Current Executing Process: 
  pid: 4170, tid: 4214
  com.serenegiant

Backtrace: 
    #00 pc 00009388  /system/lib/libandroid.so (ANativeWindow_lock+19)
    #01 pc 00010f50  /system/lib/libUVCCamera.so (_Z13copyToSurfaceP9uvc_framePP13ANativeWindow+480)
    #02 pc 000112c4  /system/lib/libUVCCamera.so (_ZN10UVCPreview18do_capture_surfaceEP7_JNIEnv+184)
    #03 pc 00011178  /system/lib/libUVCCamera.so (_ZN10UVCPreview10do_captureEP7_JNIEnv+116)
    #04 pc 00010bd4  /system/lib/libUVCCamera.so (_ZN10UVCPreview19capture_thread_funcEPv+64)
    #05 pc 0004243b  /system/lib/libc.so (_ZL15__pthread_startPv+30)
    #06 pc 000192f1  /system/lib/libc.so (__start_thread+6)

$** *** *** *** *** *** *** *** Exception *** *** *** *** *** *** *** **$

cnpalms, 2018-09-12:
Hi,Have you solved this issue?
@xiaochongdeweiba

xiaochongdeweiba, 2018-09-12:
no, but you can use other way to make a recorder, There has been a decrease on the mistake.

cnpalms, 2018-09-12:
ok,thanks.