saki4510t/UVCCamera #228: Is it possible to connect to 2 cameras in service based on test4?

lb5160482, 2017-09-01:
By using one camera, test 4 is good enough.  Test7 is able to connect to 2 cameras, but everything is in activity. Is it possible to use the mechanism in test4 to get access to 2 cameras?

jp1017, 2017-09-09:
Yes, two CameraClient is ok.

lb5160482, 2017-09-11:
@jp1017 With two servers and two services running? So I need to copy the current service and rename it. Is that correct?

jp1017, 2017-09-12:
No, you can do like this: 

```
CameraClient clinet1;
CameraClient clinet2;
```

lb5160482, 2017-09-12:
@jp1017 Thanks, I see what you mean. But what I am confused is that there is only one UVCService running, which will bind to the client. But now we need to have the clients to connect to two different devices. Are we able to do that by only running one service?

jp1017, 2017-11-14:
Hi, @lb5160482 Have you solved it?

Steven9402, 2018-06-06:
see https://github.com/Steven9402/AndroidUSBCameraInterface.git for example