saki4510t/UVCCamera #231: test7 两个USB摄像头不能同时显示

liyanmei, 2017-09-08:
一部手机上连接两个USB摄像头，一个格式是UVCCamera.FRAME_FORMAT_YUYV，另外一个格式是FRAME_FORMAT_MJPEG，目前运行test7，只有FRAME_FORMAT_MJPEG成功，FRAME_FORMAT_YUYV永远不成功。而且在test0上运行的时候，连接一个摄像头，这两个都可以实现预览的。

jp1017, 2017-09-08:
报错呢, 你的两个摄像头是连的连个usb是一个hub分出来的,还是直接连的usb host

liyanmei, 2017-09-08:
是一个hub分出去连接的两个摄像头。没有报错信息，FRAME_FORMAT_YUYV这个格式的，一直是黑屏状态。

jp1017, 2017-09-08:
只连一个yuyv也不行吗, 换个摄像头,换个手机都试试呢???

liyanmei, 2017-09-08:
只连一个yuyv，其他的项目都可以，就是test7不行

liyanmei, 2017-09-08:
09-08 10:57:29.955 1350-1350/? I/Timeline: Timeline: Activity_launch_request id:com.serenegiant.usbcameratest7 time:4613194
09-08 10:57:29.965 1310-1346/? D/SystemAdController: Package:com.serenegiant.usbcameratest7 is not in the list
09-08 10:57:29.965 1008-6036/? I/ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.serenegiant.usbcameratest7/.MainActivity bnds=[34,657][170,793]} from pid 1350
09-08 10:57:30.015 1008-1019/? I/ActivityManager: Start proc com.serenegiant.usbcameratest7 for activity com.serenegiant.usbcameratest7/.MainActivity: pid=25470 uid=10395 gids={50395, 1028, 1015}
09-08 10:57:30.135 25470-25470/? D/ActivityThread: handleBindApplication:com.serenegiant.usbcameratest7
09-08 10:57:30.165 25470-25470/? I/dalvikvm: Could not find method com.serenegiant.common.BaseActivity.requestPermissions, referenced from method com.serenegiant.common.BaseActivity.onMessageDialogResult
09-08 10:57:30.175 25470-25470/? I/dalvikvm: Could not find method android.app.Activity.onRequestPermissionsResult, referenced from method com.serenegiant.common.BaseActivity.onRequestPermissionsResult
09-08 10:57:30.295 218-1035/? D/Utils: printFileName fd(37) -> /data/app/com.serenegiant.usbcameratest7-1.apk
09-08 10:57:30.305 25470-25470/? I/dalvikvm: Could not find method android.hardware.usb.UsbDevice.getSerialNumber, referenced from method com.serenegiant.usb.USBMonitor.getDeviceKeyName
09-08 10:57:30.305 25470-25470/? I/dalvikvm: Could not find method android.hardware.usb.UsbDevice.getManufacturerName, referenced from method com.serenegiant.usb.USBMonitor.getDeviceKeyName
09-08 10:57:30.305 25470-25470/? I/dalvikvm: Could not find method android.hardware.usb.UsbDevice.getManufacturerName, referenced from method com.serenegiant.usb.USBMonitor.updateDeviceInfo
09-08 10:57:30.305 25470-25470/? I/dalvikvm: Could not find method android.hardware.usb.UsbDevice.getVersion, referenced from method com.serenegiant.usb.USBMonitor.updateDeviceInfo
09-08 10:57:30.305 218-218/? D/Utils: printFileName fd(40) -> /data/app/com.serenegiant.usbcameratest7-1.apk
09-08 10:57:30.535 1008-1022/? I/ActivityManager: Displayed com.serenegiant.usbcameratest7/.MainActivity: +532ms
09-08 10:57:30.535 1008-1022/? I/Timeline: Timeline: Activity_windows_visible id: ActivityRecord{430fdaf0 u0 com.serenegiant.usbcameratest7/.MainActivity t33} time:4613773
09-08 10:57:30.535 1008-1022/? I/WindowManager: Screen frozen for +483ms due to Window{43122d08 u0 com.serenegiant.usbcameratest7/com.serenegiant.usbcameratest7.MainActivity}
09-08 10:57:31.325 25470-25491/com.serenegiant.usbcameratest7 W/Toast: From com.serenegiant.usbcameratest7, go ahead.
09-08 10:57:31.325 25470-25491/com.serenegiant.usbcameratest7 W/Toast: From com.serenegiant.usbcameratest7, go ahead.
09-08 10:57:37.715 25470-25470/com.serenegiant.usbcameratest7 I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@4247c200 time:4620955
09-08 10:57:37.735 25470-25491/com.serenegiant.usbcameratest7 I/dalvikvm: Could not find method android.hardware.usb.UsbInterface.getAlternateSetting, referenced from method com.serenegiant.usb.USBMonitor$UsbControlBlock.getInterface
09-08 10:57:37.735 25470-25491/com.serenegiant.usbcameratest7 W/dalvikvm: VFY: unable to resolve virtual method 730: Landroid/hardware/usb/UsbInterface;.getAlternateSetting ()I
09-08 10:57:37.735 25470-25491/com.serenegiant.usbcameratest7 D/dalvikvm: VFY: replacing opcode 0x6e at 0x0039
09-08 10:57:37.755 25470-25491/com.serenegiant.usbcameratest7 D/UsbDeviceConnectionJNI: close
09-08 10:57:37.755 25470-25491/com.serenegiant.usbcameratest7 I/USBMonitor: name=/dev/bus/usb/001/011,desc=88,busnum=1,devnum=11,rawDesc=[B@42651510
09-08 10:57:37.755 25470-25491/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: getPreviewTexture:
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 V/CameraThread: handleOpen:
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 V/CameraThread: handleClose:
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 V/CameraThread: handleStopRecording:mMuxer=null
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: setVideoEncoder:
09-08 10:57:37.755 25470-25502/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: RenderThread#setEncoder:encoder=null
09-08 10:57:37.755 25470-25502/com.serenegiant.usbcameratest7 I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
09-08 10:57:37.755 25470-25502/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: updatePreviewSurface:tex_id=1
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest7-1/libjpeg-turbo1500.so 0x42482338
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest7-1/libjpeg-turbo1500.so 0x42482338
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest7-1/libjpeg-turbo1500.so 0x42482338, skipping init
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest7-1/libusb100.so 0x42482338
09-08 10:57:37.755 25470-25486/com.serenegiant.usbcameratest7 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest7-1/libusb100.so 0x42482338
[... cut ...]
09-08 10:57:44.685 25470-25489/com.serenegiant.usbcameratest7 I/AbsUVCCameraHandler: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","160x120","320x240","176x144","352x288"]}]}
09-08 10:57:44.685 25470-25489/com.serenegiant.usbcameratest7 V/CameraThread: handleStartPreview:
09-08 10:57:44.685 25470-25921/com.serenegiant.usbcameratest7 I/libUVCCamera: [25921*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
09-08 10:57:44.695 25470-25923/com.serenegiant.usbcameratest7 I/libUVCCamera: [25923*UVCPreview.cpp:258:callbackPixelFormatChanged]:PIXEL_FORMAT_RAW:

liyanmei, 2017-09-08:
FRAME_FORMAT_YUYV对应的bandwidthFactor为1.0f；
FRAME_FORMAT_MJPEG对应的bandwidthFactor为0.5f；直接对应修改就能解决黑屏的问题；

GongShaoqing, 2017-09-09:
@liyanmei  test7没有用到bandwidthFactor，你是怎么修改的，能贴一下相关的代码么？

GongShaoqing, 2017-09-11:
@liyanmei  两个相机都是MJPEG格式，但两个相机不能同时显示，只能显示一个，另外一个直接黑屏，这种情况也可以用你说的方法改么？

liyanmei, 2017-09-11:
@GongShaoqing 你可以先试试，应该是可以的

ulike, 2017-09-11:
@liyanmei 例子7里面bandwidthFactor本来就是0.5啊，目前测试打开一个没问题，同时打开两个还是报错

jp1017, 2017-09-11:
> @liyanmei 两个相机都是MJPEG格式，但两个相机不能同时显示，只能显示一个，另外一个直接黑屏，这种情况也可以用你说的方法改么？

我试过了,只能一个正常, 另一个黑屏...依旧,,仍然是带宽不足问题,,,

liyanmei, 2017-09-11:
@jp1017 @ulike 首先要确定你连接的usb相机是哪种类型的，然后再改，
FRAME_FORMAT_YUYV对应的bandwidthFactor为1.0f；
FRAME_FORMAT_MJPEG对应的bandwidthFactor为0.5f；直接对应修改就能解决黑屏的问题；
这个都是一一对应的

ulike, 2017-09-11:
@liyanmei 我的摄像头是双目摄像头，一个红外一个可将光在一个板子上，直接连的一个usb host，Android系统能识别为两个usb，用例子7跑的时候打开一个能正常打开，打开第二个时usb就dettach了，需要拔掉usb重插才能重新attach。

GongShaoqing, 2017-09-12:
@liyanmei  我下载的文件中少了不少包，如：com.serenegiant.utils，这是怎么回事?

fengbenpaao, 2017-11-07:
很明显，你上层有解码 yuyv吗？  没有的话，怎么显示呢？ 在jni源码那里  do_preview 那里， 做了分叉， 看看那里能否支持显示出 yuyv；

MrXRain, 2017-12-21:
应该是USB2.0口带宽不够，我换成USB3.0或是typeC的口后，两个摄像头都可以正常预览显示

jp1017, 2017-12-21:
更换了两个MJPEG格式的摄像头,并且分辨率改为320*240,两个可以同时预览,拍照,摄像

iflove, 2018-01-12:
@jp1017 真的可以同时预览,拍照,摄像？

jp1017, 2018-01-15:
@iflove 可以啊，买摄像头时买MJPEG格式输出的

iflove, 2018-02-08:
是可以，换两个一百万就可以了，不然带宽不足

ChenArno, 2018-04-22:
我这边试了320*240的，是可以同时预览，拍照，有木有什么通过硬件或者软件的解决办法，提升分辨率呢。谢谢

iflove, 2018-07-03:
@ChenArno 是否报宽带不足

ChenArno, 2018-07-03:
@iflove 我改了一个参数，可以达到640*480，```Java     private float banwidth = 0.65f;//双摄像头，0.65f  1f //最大0.75f  ```

qyd9212, 2020-06-18:
> @iflove 我改了一个参数，可以达到640*480，`Java private float banwidth = 0.65f;//双摄像头，0.65f 1f //最大0.75f `

请问您有demo可以借鉴一下吗

qyd9212, 2020-06-18:
> 更换了两个MJPEG格式的摄像头,并且分辨率改为320*240,两个可以同时预览,拍照,摄像

请问您有demo可以借鉴一下吗